acecilia/OpenWRTInvasion #3: ac2100

kapook007, 2020-03-31:
root@ubuntu:~/OpenWRTInvasion-master# python3 remote_command_execution_vulnerability.py
Router IP address: 192.168.31.1
stok: d6aad3bd00d6ef3b7d7d82f9b0817ad6
****************
router_ip_address: 192.168.31.1
stok: d6aad3bd00d6ef3b7d7d82f9b0800ad6
****************
start uploading config file...
start exec command...
done! Now you can connect to the router using telnet (user: root, password: none)
In MacOS, execute in the terminal:
telnet 192.168.31.1
root@ubuntu:~/OpenWRTInvasion-master#

i can not telnet to ac2100
ac2100 no telnet server

acecilia, 2020-03-31:
Then maybe the exploit is failing. I can’t help you, I don’t have an ac2100 router to test on

Dreamoffice, 2020-03-31:
i have the exact same problem with the Mi Router 4A Giga Version. Firmware 2.28.132

acecilia, 2020-03-31:
@Dreamoffice check your network configuration or use version `0.0.1`

acecilia, 2020-03-31:
Also, the exploit on version `0.0.2` (telnet shell) requires the router to have access to the internet

Dreamoffice, 2020-03-31:
the router is setup as a repeater. The WAN of the router is connected to my LAN. 
The router has internet access. i checked the Telnet connection with LAN and wifi. but still refusing.

Dreamoffice, 2020-03-31:
i did version 0.0.1 with a raspi 4
version 0.0.2 is not running on my mac.

closed 2020-04-02 by acecilia