acecilia/OpenWRTInvasion #5: No work

SergeyDjam, 2020-04-05:
Hello

Mi Router 4A Giga Version | MiWiFi Release 2.28.132

1. git clone
2. python3 remote_command_execution_vulnerability.py
`Router IP address: 192.168.31.1
stok: 123123123
****************
router_ip_address: 192.168.31.1
stok: 123123123
****************
start uploading config file...
start exec command...
done! Now you can connect to the router using telnet (user: root, password: none)
In MacOS, execute in the terminal:
telnet 192.168.31.1
`
3. telnet 192.168.31.1
Trying 192.168.31.1...
telnet: connect to address 192.168.31.1: Connection refused

SergeyDjam, 2020-04-05:
fix it. On Internet

closed 2020-04-05 by SergeyDjam

johol, 2020-04-05:
I have the same problem. What have you done to fix it?

acecilia, 2020-04-05:
@johol the router needs internet access

johol, 2020-04-05:
Thanks, after rebooting the router everything is working.