acecilia/OpenWRTInvasion #21: Thanks, applicable in mi router 4 version 2.26.175

xy9860, 2020-06-22:
Use v0.01 成功
Use v0.02 连接拒绝

acecilia, 2020-06-22:
Thanks for reporting it :)

closed 2020-06-22 by acecilia

Firef0x, 2020-09-21:
@acecilia Maybe it's better to add a note in README.md for the correct version applicable in Mi router 4? I found the correct version here but it's not easy to find.

acecilia, 2020-09-21:
In the readme it says:

`Mi Router 4Q (aka R4C): user cadaverous claims that this method also works on firmware version 2.28.48 (message posted in Slack), but because the router is mips architecture (not mipsel), he needed to use version 0.0.1 of the script (the other versions use a busybox binary built for the mipsel architecture that is used to start a telnet sever).`

Isnt that enough? What is exactly your router version? `4C`? `4Q`?

Firef0x, 2020-10-15:
@acecilia It's `Mi Router 4` (No suffix): https://www.mi.com/miwifi4
Is it covered in this repository?

acecilia, 2020-10-17:
I dont know. If you try and it works, please report the router version and the software version so I can add it to the readme :)

juampe, 2020-12-02:
I tried to root the R4 version, but was unsuccesfull with 2.26.175 firmare version and v0.0.1

miwifi_r4_firmware_8ed47_2.26.175.bin
https://mirom.ezbox.idv.tw/en/miwifi/R4/

```
root@controller:/ins/OpenWRTInvasion-0.0.1# python3 remote_command_execution_vulnerability.py
Start netcat on port 4444
(The way to do this in MacOS is to open a terminal and run '/usr/bin/nc -l 4444')
When you are done, press any key to continue
Router IP address: 192.168.0.6
Your IP address: 192.168.0.1
stok: aa364a4225c326695e59ebcb6ba4901d
****************
netcat_port: 4444
attacker_ip_address: 192.168.0.1
router_ip_address: 192.168.0.6
stok:aa364a4225c326695e59ebcb6ba4901d
****************
start uploading config file ...
start exec command...
done!
```

Firef0x, 2020-12-20:
> I dont know. If you try and it works, please report the router version and the software version so I can add it to the readme :)

@acecilia Worked on [Mi Router 4](https://www.mi.com/miwifi4) with firmware version `v2.26.175` and [OpenWRTInvasion v0.0.1](https://github.com/acecilia/OpenWRTInvasion/tree/0.0.1).

acecilia, 2020-12-20:
@Firef0x thanks, added to the README 🙂

EmericLee, 2021-10-06:
Thanks! Succesfully use the lasted release V0.07 on  Mi Router 4 (R4)  with firmware version v2.26.175.

NOT: THE ROUTER NEEDS INTERNET ACCESS.  
The router interface wan must connected with internet. otherwsie root will failed without any tips.

emeric254, 2021-10-06:
> @acecilia Worked on [Mi Router 4](https://www.mi.com/miwifi4) with firmware version `v2.26.175` and [OpenWRTInvasion v0.0.1](https://github.com/acecilia/OpenWRTInvasion/tree/0.0.1).

@acecilia First thanks a lot ! This exploit works very well for miwifi r4a 100m and miwifi r4.
You might want to write that the router need to be setup as a router, not as an wifi access point. The explanation is that in wifi access point the web-ui does not give access to all the features like the speedtest etc..

minikolic, 2022-12-15:
`v0.0.10` works with `v2.26.175`! The only problem that I faced is that OpenWRTInvasion is not working on WSL (Windows Subsystem for Linux) so I had to spin up Linux from a stick

JiinJie, 2022-12-30:
mi router 4: version 2.26.175. 

I was used the script version v0.0.1 and v0.0.10 ,all of them dosen't work , the route can connect to the Inernet

lkpopo, 2023-01-11:
> In the readme it says:
>
> `Mi Router 4Q (aka R4C): user cadaverous claims that this method also works on firmware version 2.28.48 (message posted in Slack), but because the router is mips architecture [...]

C:\Users\Jack Deng>nc -l 4444
local listen fuxored: INVAL
when i open another terminal ,run the nc -l 4444,this happened.sys:win 10 how can i solve this problem.

JiinJie, 2023-01-12:
> 

the script is not support native windows system you have to use linux or macos .another way is using docker to run images on windows。please read the README.md

HeJerry, 2023-06-20:
I try use docerk run code,but it's seem still not work
C:\Users\Jerry>docker run --network host -it openwrtinvasion
Router IP address [press enter for using the default 'miwifi.com']: 192.168.31.1
Enter router admin password: www.ebep.com
There two options to provide the files needed for invasion:
   1. Use a local TCP file server runing on random port to provide files in local directory `script_tools`.
   2. Download needed files from remote github repository. (choose this option only if github is accessable inside router device.)
Which option do you prefer? (default: 1)
****************
router_ip_address: 192.168.31.1
stok: e2c4720de879957fa8f85fc64c13c691
file provider: local file server
****************
start uploading config file...
start exec command...
local file server is runing on 0.0.0.0:44357. root='script_tools'
Warning: the process has finished, but seems like ssh connection to the router is not working as expected.
* Maybe your firmware version is not supported, please have a look at https://github.com/acecilia/OpenWRTInvasion/blob/master/README.md#unsupported-routers-and-firmware-versions
* Anyway you can try it with: telnet 192.168.31.1

try use telnet 192.168.31.1 not work

white-z, 2024-09-16:
> I try use docerk run code,but it's seem still not work C:\Users\Jerry>docker run --network host -it openwrtinvasion Router IP address [press enter for using the default 'miwifi.com']: 192.168.31.1 [...]

请问你解决了吗？我现在遇到同样的问题，一直卡在这里，无法链接telnet