acecilia/OpenWRTInvasion #23: Lici install problem

BosakS, 2020-07-20:
root@OpenWrt:~# opkg update
Downloading http://downloads.openwrt.org/snapshots/targets/ramips/mt7621/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading http://downloads.openwrt.org/snapshots/targets/ramips/mt7621/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
root@OpenWrt:~# opkg install luci
Installing luci (git-20.200.48516-dcac704) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/luci/luci_git-20.200.48516-dcac704_all.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci:
 *      kernel (= 5.4.52-1-81cda4f04181e61954d91615ef38462c)
 * opkg_install_cmd: Cannot install package luci.
root@OpenWrt:~# uname -a
Linux OpenWrt 5.4.43 #0 SMP Fri Jun 5 20:03:26 2020 mips GNU/Linux

closed 2020-07-20 by acecilia

Diablofx, 2020-07-27:
Hello How did you solved the issue ?

BosakS, 2020-07-27:
I used another firmware containing Luci.  Now everything works.