acecilia/OpenWRTInvasion #27: Can't use telnet to write.

zy-zero, 2020-08-28:
Hi, it looks like can't use telnet with windows to 4a Gigabit. Why? Do I need use linux to try? and the python3 is required?

zy-zero, 2020-08-28:
Can't use telnet with python3, and python 2.7, not sure where I am wrong.

acecilia, 2020-08-28:
I used worked on the code in Mac. Some other people tried on linux. A very few mentioned windows.

Answering your questions:

> Why?

No idea, I do not use windows.

> Do I need use linux to try?

Most probably, it will be easier with linux or mac.

> and the python3 is required?

Yes.

Please search for a solution on the OpenWrt forum (most probably somebody else has been through a windows setup), or ask there: https://forum.openwrt.org/t/xiaomi-mi-router-4a-gigabit-edition-r4ag-r4a-gigabit-fully-supported-and-flashable-with-openwrtinvasion

closed 2020-08-28 by acecilia