acecilia/OpenWRTInvasion #30: <help needed> did i brick my router？

zhuo1angT, 2020-09-01:
following the directions in the README.md, i got into the router and downloaded the firmware image, checked the md5 and its correct.
while installing the OpenWrt, the CLI prints 'rebooting' in the end, and the connection was done.
there is no wifi, and wired connection to LAN port is also unavialeable.
i thought i had did it right, could you give me some instructions?

zhuo1angT, 2020-09-01:
also the light on the router is orange.
my router is Xiaomi R4A
org firmware version 2.28.61
using the openwrt image provided by the repo owner @acecilia

acecilia, 2020-09-01:
I cant help you much more, all I know is in the readme. Check the readme for instructions on what to do if you brick your router, or ask in the OpenWrt forum: https://forum.openwrt.org/t/xiaomi-mi-router-4a-gigabit-edition-r4ag-r4a-gigabit-fully-supported-and-flashable-with-openwrtinvasion

closed 2020-09-01 by acecilia