acecilia/OpenWRTInvasion #31: Router cant uncompress files

Jiaocz, 2020-09-04:
When it uploaded config files (payload.tar.gz), it said "code":"1629:, msg is Error uncompressingm, the fies may damaged.
[image]

acecilia, 2020-09-04:
I think that message is received every time, the exploit works anyway

closed 2020-09-04 by acecilia

Jiaocz, 2020-09-04:
I found the reason, it is because that in the speedtest, some url can be accessed in my area, but if I replace some of these, then the message will become "Error speedtesting" but not Script executed.

acecilia, 2020-09-04:
Make sure your network configuration is as simple as possible, that you are configuring the right IPs and that connection between your computer and the router is happening

hstr0100, 2020-09-10:
Not sure where to put this comment, but it should help whoever has the same problem as me:

On my 4A, versions 0.0.2 and 0.0.3 of the exploit did not work. the commands did get executed on the router but the telnet daemon did not work.
I managed to gain shell access on the firmware 2.28.132 by using netcat (version 0.0.1 of this exploit).

From there, I manually executed the exploit command and got the output "curl (51) error"
This means that for some reason my router wasn't able to verify the SSL certificate of the busybox download site, so from there everything failed.
by slapping --insecure to the curl command string I was able to successfully run the exploit v0.0.3.

So if you run into the same problem as me, append `--insecure` to https://github.com/acecilia/OpenWRTInvasion/blob/master/script.sh#L17 and have fun.
Some may critique the fact that we're ignoring the SSL certificate there, and that could be potentially dangerous, but this is no worse than "curl-ing" an insecure http:// url anyway, and there's not much to do to fix this problem anyway aside from including busybox into the payload I guess.

By doing this you also don't have to reset your router nor anything. it seems that this is the cause of most of the issues previously related here. so I hope this helps so we can get more people onboard custom firmwares for this router.

acecilia, 2020-09-11:
@hstr0100 nice! Ill include `--insecure` in the script. Thanks!

acecilia, 2020-09-11:
Also released new version 0.0.4 😄  https://github.com/acecilia/OpenWRTInvasion/releases/tag/0.0.4