acecilia/OpenWRTInvasion #33: MI router 4 - stok versions

coinsbarboss, 2020-09-07:
Exploit script stop working after updating the router.
The last firmware version seems to be: 2.26.134

I think the problem is that I'm using the old stok version in script...but unfortunately I can't check the correct version myself.

Is it possible to get the list of possible stok versions on official forums, etc?

coinsbarboss, 2020-09-07:
` python3 remote_command_execution_vulnerability.py
****************
router_ip_address: 192.168.31.1
stok: a5445597af88bf26af058d7098e67208
****************
start uploading config file...
start exec command...
done! Now you can connect to the router using telnet (user: root, password: none)
In MacOS, execute in the terminal:
telnet 192.168.31.1
root@worktime:~/OpenWRTInvasion# telnet 192.168.31.1
Trying 192.168.31.1...
telnet: Unable to connect to remote host: Connection refused`

acecilia, 2020-09-07:
I am not sure if the exploit is compatible with the new firmware `3.0.24`. I asked [here](https://forum.openwrt.org/t/xiaomi-mi-router-4a-gigabit-edition-r4ag-r4a-gigabit-fully-supported-and-flashable-with-openwrtinvasion/36685/1122). Please report if you manage to make it work :)

coinsbarboss, 2020-09-07:
> I am not sure if the exploit is compatible with the new firmware `3.0.24`. I asked [here](https://forum.openwrt.org/t/xiaomi-mi-router-4a-gigabit-edition-r4ag-r4a-gigabit-fully-supported-and-flashable-with-openwrtinvasion/36685/1122). Please report if you manage to make it work :)

Actually I don't think that firmware is 3.0.24 ver.

And my question was about "stok version hashes". If I understand correctly each firmware ver has it's own stok hash. Right?
If I will use incorrect stok hash, it will not work. Correct?

Is there any database with stok hashes depending on firmware versions?
 For example: If I need to know a stok hash for firmware 2.26.134?

acecilia, 2020-09-07:
No. You get a new STOK every time you login you the router using the web interface. You get the STOK from the url. See the readme, it is explained there in a big picture with a red square.

closed 2020-09-07 by acecilia

coinsbarboss, 2020-09-08:
So if I don't know the password from 192.168.31.1 - it's not possible to use the hack?

> No. You get a new STOK every time you login you the router using the web interface. You get the STOK from the url. See the readme, it is explained there in a big picture with a red square.

acecilia, 2020-09-08:
You are correct, it’s not possible to hack the router without knowing the password

komplykated, 2022-09-22:
> You are correct, it’s not possible to hack the router without knowing the password

Hi acecilia, in my case, it did not asked the stok, and auto generate the different stok compare to the one from the web browser. What can I do in this case please?