acecilia/OpenWRTInvasion #34: r1 code=1629

z1001001100, 2020-09-07:
when i exec  "remote_command_execution_vulnerability.py", it told "done". but I still can't telnet into my router.
after that I find result  in upload config file part said: 
`“ start uploading config file...
r1={"code":1629,"msg":"解压失败，可能文件已经损坏"}” `
It mean unzip fail, file may destoryed.
what should i do?

z1001001100, 2020-09-07:
i change the code in the  "remote_command_execution_vulnerability.py" like that
`## upload config file
print("start uploading config file...")
r1 = requests.post("http://{}/cgi-bin/luci/;stok={}/api/misystem/c_upload".format(router_ip_address, stok), files={"image":open("build/payload.tar.gz",'rb')}, proxies=proxies)
print("r1="+r1.text)`
and i find the error

acecilia, 2020-09-07:
Please look at the closed issues before opening a new one. This has already been discussed: https://github.com/acecilia/OpenWRTInvasion/issues/31

closed 2020-09-07 by acecilia