acecilia/OpenWRTInvasion #36: a lot errors when run pip3 install -r requirements.txt on my macOS 10.15.7

iovoz, 2020-09-27:
Last login: Mon Sep 28 03:29:02 on console
riovo@Rs-MacBook-Pro ~ % python3
Python 3.8.2 (default, Aug 25 2020, 09:23:57) 
[Clang 12.0.0 (clang-1200.0.32.2)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> 
KeyboardInterrupt
>>> ^D
riovo@Rs-MacBook-Pro ~ % pip3 --version
pip 19.2.3 from /Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/site-packages/pip (python 3.8)
riovo@Rs-MacBook-Pro ~ % telnet
telnet> %                                                                                                         
riovo@Rs-MacBook-Pro ~ % cd Desktop/OpenWRTInvasion 
riovo@Rs-MacBook-Pro OpenWRTInvasion % pwd
/Users/riovo/Desktop/OpenWRTInvasion
riovo@Rs-MacBook-Pro OpenWRTInvasion % ls -all
total 104
drwxr-xr-x@ 15 riovo  staff   480 Sep 28 02:18 .
drwx------@  6 riovo  staff   192 Sep 28 03:00 ..
-rw-r--r--@  1 riovo  staff  8196 Sep 28 02:30 .DS_Store
drwxr-xr-x@ 12 riovo  staff   384 Sep 28 02:17 .git
-rw-r--r--   1 riovo  staff     7 Sep 28 02:16 .gitignore
-rw-r--r--@  1 riovo  staff  8575 Sep 28 02:16 README.md
drwxr-xr-x@  6 riovo  staff   192 Sep 28 02:17 extras
drwxr-xr-x@  4 riovo  staff   128 Sep 28 02:17 firmwares
drwxr-xr-x@  5 riovo  staff   160 Sep 28 02:16 readme
-rw-r--r--@  1 riovo  staff   290 Sep 28 02:18 readme-roy.txt
-rw-r--r--@  1 riovo  staff  2852 Sep 28 02:16 remote_command_execution_vulnerability.py
-rw-r--r--@  1 riovo  staff    18 Sep 28 02:16 requirements.txt
-rw-r--r--@  1 riovo  staff  1996 Sep 28 02:16 script.sh
-rw-r--r--@  1 riovo  staff   838 Sep 28 02:16 set_english.py
-rw-r--r--@  1 riovo  staff  1856 Sep 28 02:16 speedtest_urls_template.xml
riovo@Rs-MacBook-Pro OpenWRTInvasion % pip3 install -r requirements.txt
Collecting pycrypto (from -r requirements.txt (line 1))
  Using cached https://files.pythonhosted.org/packages/60/db/645aa9af249f059cc3a368b118de33889219e0362141e75d4eaf6f80f163/pycrypto-2.6.1.tar.gz
Collecting requests (from -r requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/45/1e/0c169c6a5381e241ba7404532c16a21d86ab872c9bed8bdcd4c423954103/requests-2.24.0-py2.py3-none-any.whl
Collecting certifi>=2017.4.17 (from requests->-r requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/5e/c4/6c4fe722df5343c33226f0b4e0bb042e4dc13483228b4718baf286f86d87/certifi-2020.6.20-py2.py3-none-any.whl
Collecting urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 (from requests->-r requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/9f/f0/a391d1463ebb1b233795cabfc0ef38d3db4442339de68f847026199e69d7/urllib3-1.25.10-py2.py3-none-any.whl
Collecting idna<3,>=2.5 (from requests->-r requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/a2/38/928ddce2273eaa564f6f50de919327bf3a00f091b5baba8dfa9460f3a8a8/idna-2.10-py2.py3-none-any.whl
Collecting chardet<4,>=3.0.2 (from requests->-r requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/bc/a9/01ffebfb562e4274b6487b4bb1ddec7ca55ec7510b22e4c51f14098443b8/chardet-3.0.4-py2.py3-none-any.whl
Building wheels for collected packages: pycrypto
  Building wheel for pycrypto (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /Library/Developer/CommandLineTools/usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/xd/c1h_3xmj12q08slfnzr9v8fr0000gn/T/pip-install-p9qp1zro/pycrypto/setup.py'"'"'; __file__='"'"'/private/var/folders/xd/c1h_3xmj12q08slfnzr9v8fr0000gn/T/pip-install-p9qp1zro/pycrypto/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/xd/c1h_3xmj12q08slfnzr9v8fr0000gn/T/pip-wheel-beiq7txo --python-tag cp38
       cwd: /private/var/folders/xd/c1h_3xmj12q08slfnzr9v8fr0000gn/T/pip-install-p9qp1zro/pycrypto/
  Complete output (299 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.macosx-10.14.6-x86_64-3.8
  creating build/lib.macosx-10.14.6-x86_64-3.8/Crypto
  copying lib/Crypto/pct_warnings.py -> build/lib.macosx-10.14.6-x86_64-3.8/Crypto
  copying lib/Crypto/__init__.py -> build/lib.macosx-10.14.6-x86_64-3.8/Crypto
  creating build/lib.macosx-10.14.6-x86_64-3.8/Crypto/Hash
  copying lib/Crypto/Hash/MD2.py -> build/lib.macosx-10.14.6-x86_64-3.8/Crypto/Hash
  copying lib/Crypto/Hash/SHA512.py -> build/lib.macosx-10.14.6-x86_64-3.8/Crypto/Hash
  copying lib/Crypto/Hash/SHA384.py -> build/lib.macosx-10.14.6-x86_64-3.8/Crypto/Hash
  copying lib/Crypto/Hash/SHA256.py -> build/lib.macosx-10.14.6-x86_64-3.8/Crypto/Hash
  copying lib/Crypto/Hash/hashalgo.py -> build/lib.macosx-10.14.6-x86_64-3.8/Crypto/Hash
  copying lib/Crypto/Hash/__init__.py -> build/lib.macosx-10.14.6-x86_64-3.8/Crypto/Hash
  copying lib/Crypto/Hash/SHA.py -> build/lib.macosx-10.14.6-x86_64-3.8/Crypto/Hash
  copying lib/Crypto/Hash/SHA224.py -> build/lib.macosx-10.14.6-x86_64-3.8/Crypto/Hash
  copying lib/Crypto/Hash/MD4.py -> build/lib.macosx-10.14.6-x86_64-3.8/Crypto/Hash
  copying lib/Crypto/Hash/HMAC.py -> build/lib.macosx-10.14.6-x86_64-3.8/Crypto/Hash
[... 514 lines cut ...]
    In file included from src/MD2.c:30:
    In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/string.h:61:
    /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/_types.h:43:9: error: unknown type name '__uint32_t'; did you mean '__uint128_t'?
    typedef __uint32_t      __darwin_wctype_t;
            ^
    note: '__uint128_t' declared here
    In file included from src/MD2.c:30:
    In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/string.h:64:
    /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/sys/_types/_size_t.h:31:9: error: unknown type name '__darwin_size_t'
    typedef __darwin_size_t        size_t;
            ^
    In file included from src/MD2.c:30:
    In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/string.h:152:
    In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/sys/_types/_ssize_t.h:30:
    /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/machine/types.h:37:2: error: architecture not supported
    #error architecture not supported
     ^
    fatal error: too many errors emitted, stopping now [-ferror-limit=]
    20 errors generated.
    error: command 'xcrun' failed with exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /Library/Developer/CommandLineTools/usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/xd/c1h_3xmj12q08slfnzr9v8fr0000gn/T/pip-install-p9qp1zro/pycrypto/setup.py'"'"'; __file__='"'"'/private/var/folders/xd/c1h_3xmj12q08slfnzr9v8fr0000gn/T/pip-install-p9qp1zro/pycrypto/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/xd/c1h_3xmj12q08slfnzr9v8fr0000gn/T/pip-record-y1dc7wiu/install-record.txt --single-version-externally-managed --compile Check the logs for full command output.
WARNING: You are using pip version 19.2.3, however version 20.2.3 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
riovo@Rs-MacBook-Pro OpenWRTInvasion %

acecilia, 2020-09-27:
Your computer setup is not something I can help you with. Closing

closed 2020-09-27 by acecilia

iovoz, 2020-09-27:
seems pycrypto not compiled with  python3. do you think use pycryptodome would better?

iovoz, 2020-09-27:
> Your computer setup is not something I can help you with. Closing

seems pycrypto not compiled with python3. do you think use pycryptodome would better?

acecilia, 2020-09-27:
I don’t know, I’m have very limited knowledge of the python ecosystem

iovoz, 2020-09-28:
I got the same errors in python 3.6 with pycrypto on my CentOS . Then use pycryptohome fixed it. Also fixed on my MacOS which using python 3.8. Have been looking through some documents, pycrypto not maintained a long time and recommended to pycryptohome. Can I guess you don’t have issue on you environment because it is not updated?

acecilia, 2020-09-28:
Could be, I don’t know. Feel free to open a pr