acecilia/OpenWRTInvasion #38: download fail

Henry3gs, 2020-09-30:
Anybody encounter downloading problem when using command git clone https://github.com/acecilia/OpenWRTInvasion.git to get the package?

my error message is:

error: RPC failed; curl 18 transfer closed with outstanding read data remaining
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed

It looks like one of the large file is corrupt on server. I tried it many times, never success in downloading this Invasion package.

Henry3gs, 2020-09-30:
My download stopped here: 24% (47/195)

mneamtu, 2020-10-01:
Checked today, it is working, maybe it's your side problem:

$ git clone https://github.com/acecilia/OpenWRTInvasion.git
Cloning into 'OpenWRTInvasion'...
remote: Enumerating objects: 74, done.
remote: Counting objects: 100% (74/74), done.
remote: Compressing objects: 100% (62/62), done.
remote: Total 195 (delta 33), reused 37 (delta 10), pack-reused 121
Receiving objects: 100% (195/195), 65.64 MiB | 20.94 MiB/s, done.
Resolving deltas: 100% (80/80), done.
Checking connectivity... done.

acecilia, 2020-10-01:
Thanks @mitza97

closed 2020-10-01 by acecilia