acecilia/OpenWRTInvasion #44: Unable to

dmiquel, 2020-10-31:


acecilia, 2020-10-31:
Hi 👋 

Everything I can say is already explained in the readme. 

* Try using version `0.0.1` of this exploit in case your router does not have internet access
* Try to use a simple network configuration: for example, use a normal laptop in a home network. I dont think using a raspberry is very common, and I am not sure anybody can give you support for that kind of environment.
* Last resource, read in the OpenWrt forum, there is an almost infinite thread about the router and the exploit.

closed 2020-10-31 by acecilia

dmiquel, 2020-10-31:
I tried with 0.0.1 and it did not work, turns out all these exploits only work when the router is in router mode, not when its in repeater mode. After changing this, I got in at the first try.

Thanks for the exploit!

acecilia, 2020-10-31:
I do have my router in repeater mode (4Agigabit) and exploit it without issues, so something else may be the reason preventing you from exploiting. But glad you managed 👏