acecilia/OpenWRTInvasion #46: tested on the AC2350 AIOT

jww-cw, 2020-11-26:
Unfortunately this did not work on the AC2350 AIOT  https://www.mi.com/global/mi-aiot-router-ac2350/

```
prime@ubuntu:/tmp/OpenWRTInvasion$ python3 remote_command_execution_vulnerability.py
Router IP address [press enter for using the default 192.168.31.1]: 192.168.1.131
stok: XXXXX
****************
router_ip_address: 192.168.1.131
stok: XXXXX
****************
start uploading config file...
start exec command...
done! Now you can connect to the router using several options: (user: root, password: root)
* telnet 192.168.1.131
* ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -c 3des-cbc -o UserKnownHostsFile=/dev/null root@192.168.1.131
* ftp: using a program like cyberduck
prime@ubuntu:/tmp/OpenWRTInvasion$ telnet 192.168.1.131
Trying 192.168.1.131...
telnet: Unable to connect to remote host: Connection refused
prime@ubuntu:/tmp/OpenWRTInvasion$ ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -c 3des-cbc -o UserKnownHostsFile=/dev/null root@192.168.1.131
ssh: connect to host 192.168.1.131 port 22: Connection refused
prime@ubuntu:/tmp/OpenWRTInvasion$
```

jww-cw, 2020-11-26:
FW version "version":"3.0.36"

acecilia, 2020-11-28:
Thanks for reporting it 👍 I will update the readme

closed 2020-11-28 by acecilia

eisaev, 2021-01-03:
Unfortunately, so far on this model, the only working method requires a second access point with the ability to respond to an http request. I used what was at hand: a very old ASUS WL-500gP with Entware firmware and nginx. But on a router with OpenWRT, this is even easier. See details here: https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-ax3600/55049/766

dobosz23, 2021-01-08:
Good evening
I also have a Xiaomi AIoT AC2350 router and I'm trying to connect via SSH to upload the Padavan software. I tried to start by connecting through the DAP-1635 extedner but I don't know what I'm doing wrong because it still doesn't work. Can I count on a more detailed instruction what to do?

eisaev, 2021-01-09:
@dobosz23, I am not sure that this discussion is appropriate here, because the methods that I will describe cannot be implemented in OpenWRTInvasion, because they require physical manipulation. I suggest moving the discussion of methods to a topic on the OpenWRT forum: https://forum.openwrt.org/t/support-aiot-ac2350-xiaomi/70451/26

dobosz23, 2021-01-09:
Of course, thank you for the quick reply

dobosz23, 2021-02-07:
I run on the firmware 1.3.8 CN an succesfully open SSH connection:
https://forum.openwrt.org/uploads/default/original/3X/0/5/052424e33141929b074afdf4bab9328eb8020c07.png

eisaev, 2021-11-02:
> I run on the firmware 1.3.8 CN an succesfully open SSH connection:
> 
> [image]
> 
> I run command as root and connect not a: ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -c 3des-cbc -o UserKnownHostsFile=/dev/null root@192.168.0.171 but: ssh root@192.168.0.171

It was a mistake. SSH access was obtained by a different method.  Reproducing using OpenWRTInvasion failed:
https://forum.openwrt.org/t/support-aiot-ac2350-xiaomi/70451/86