acecilia/OpenWRTInvasion #48: an UnicodeEncodeError on my environment

keke1023, 2020-11-30:
Traceback (most recent call last):
  File "remote_command_execution_vulnerability.py", line 42, in <module>
    template = f.read()
UnicodeDecodeError: 'gbk' codec can't decode byte 0xbf in position 2: illegal multibyte sequence

Traceback (most recent call last):
  File "remote_command_execution_vulnerability.py", line 46, in <module>
    f.write(data)
UnicodeEncodeError: 'gbk' codec can't encode character '\ufeff' in position 0: illegal multibyte sequence

and i add encoding='UTF-8' to open(), it works
# make config file
speed_test_filename = "speedtest_urls.xml"
with open("speedtest_urls_template.xml", "rt", encoding='UTF-8') as f:
    template = f.read()
data = template.format(router_ip_address=router_ip_address, command=command)
# print(data)
with open("build/speedtest_urls.xml", 'wt', encoding='UTF-8') as f:
    f.write(data)

acecilia, 2020-11-30:
Related: https://github.com/acecilia/OpenWRTInvasion/issues/47

Thanks! Could you please open a Pull Request with the change? 🙏

acecilia, 2020-12-02:
Fixed

closed 2020-12-02 by acecilia

keke1023, 2020-12-11:
@acecilia sorry i havent notice notification of your reply,thanks for fix it ,have a good day