acecilia/OpenWRTInvasion #49: Trying root  Mi Router 4 (R4)

juampe, 2020-12-02:
I tried to root the R4 version, but was unsuccesfull with 2.26.175 firmare version.
```
miwifi_r4_firmware_8ed47_2.26.175.bin
https://mirom.ezbox.idv.tw/en/miwifi/R4/
```
Any clues?

juampe, 2020-12-02:
As #21 with 0.0.1 but no success ¿maybe the 192.168.31.1 address is obligatory?

```
root@controller:/ins/OpenWRTInvasion-0.0.1# python3 remote_command_execution_vulnerability.py
Start netcat on port 4444
(The way to do this in MacOS is to open a terminal and run '/usr/bin/nc -l 4444')
When you are done, press any key to continue
Router IP address: 192.168.0.6
Your IP address: 192.168.0.1
stok: aa364a4225c326695e59ebcb6ba4901d
****************
netcat_port: 4444
attacker_ip_address: 192.168.0.1
router_ip_address: 192.168.0.6
stok:aa364a4225c326695e59ebcb6ba4901d
****************
start uploading config file ...
start exec command...
done!
```

acecilia, 2020-12-02:
No idea, I dont have a R4 router to test, so I cant tell if the exploit works or not 😞 

Did you run `/usr/bin/nc -l 4444` before running `python3 remote_command_execution_vulnerability.py`?
I dont know about your network configuration, but I would try to keep it as simple as possible.
You can also ask in the OpenWrt forum, there is an extremely long thread about this: https://forum.openwrt.org/t/xiaomi-mi-router-4a-gigabit-edition-r4ag-r4a-gigabit-fully-supported-and-flashable-with-openwrtinvasion/36685

acecilia, 2020-12-02:
I can see in the information you posted that the value for `stok` is `aa364a4225`. That is not a valid `stok`. Take a look at the README, where it is explained how to get the `stok`

closed 2020-12-02 by acecilia