acecilia/OpenWRTInvasion #50: Everything seems working, but I can't access Telnet / FTP / SSH

Zarto, 2020-12-03:
Hi ! Thank you for your hard work.

I just bought a Mi Router A4 100M (Model R4AC).

I applied your script to it, everyting went well, the script said "done !", but I was still unable to connect to Telnet / FTP / SSH.
Then I noticed the 3.0.5 firmware was installed on the router.

The readme says that some people managed to get it working using 2.18.58 , so I successfully downgraded to 2.18.58.
The interface is now in chinese, but that's not a problem for me.

So I applied again the script, it said "done !" but still no access. Telnet and FTP don't ask for any credentials, they just say "Connection refused".

Because I launched the script from Windows I installed Ubuntu on a computer, followed step by step this tutorial : https://www.youtube.com/watch?v=VxzEvdDWU_s but still no luck.

Can you guess what I'm doing wrong ? Is there something I can do to help you help me ?

hoddy, 2020-12-03:
It's unlikely to have been the problem both times but just to be sure did you get a new stok code when using Ubuntu and was it generated from the browser in Ubuntu?

Zarto, 2020-12-03:
Hi thank you for answering. Yes I checked and renew the stok code several time, and used it right after it was generated (didn't had the time to expire).

hoddy, 2020-12-03:
Might be worth giving version 0.0.1 a try, as it doesn't require internet access. I know you have said it has net access but just in case you're geo-blocked or the site is down that it uses to run the exploit (its a speed test server):

https://github.com/acecilia/OpenWRTInvasion/releases/tag/0.0.1

Zarto, 2020-12-03:
Just in case, can you give me an url I could try to access from a browser, to see if I get what is expected ?

I found this URL in speedtest_urls_template.xml : http://dl.ijinshan.com/safe/speedtest/FDFD1EF75569104A8DB823E08D06C21C.dat
When I try to access it from Firefox I get an "ERROR : ACCESS DENIED"

I also removed the # before print(r2.text) after the request is sent to the router by the script (just before the "done!" message).
This is what I get : {"download":1.58,"bandwidth":0.01,"code":0}

Zarto, 2020-12-03:
I managed to access command line using Netcat, I'm flashing right now.

Zarto, 2020-12-04:
OpenWRT up and running !
Many thanks for your help @hoddy hoddy, and great work @acecilia 

Sadly we still don't know why last version didn't work ...

sijanec, 2020-12-04:
Yup. Same issue with my router. I am on preloaded Chinese version 2.14.87 and I can confirm that the exploit works on OpenWRTInvasion version 0.0.1, but not on any newer version.

Regards!

acecilia, 2020-12-06:
It could potentially be that your network setup prevents the router from hitting the internet, which is required to download some binaries.

Thanks @hoddy for the support 🔥 

Closing this, as seems that using versino `0.0.1` solves the issue

closed 2020-12-06 by acecilia

Zarto, 2020-12-07:
I just want you to know this is probably not related to my network setup.
I have a "clean" access to the internet and a very basic setup (internet router with no firewall connected to the WAN port of the Xiaomi router).
It seems confirmed by @sijanec having the same trouble.
Anyway thank you again everyone !