acecilia/OpenWRTInvasion #53: Xiaomi 4A Gigabit cannot telnet.

primelyw, 2020-12-15:
~~~shell
python3 remote_command_execution_vulnerability.py
Router IP address [press enter for using the default 192.168.31.1]:
stok: correct_stok
****************
router_ip_address: 192.168.31.1
stok: correct_stok
****************
start uploading config file...
start exec command...
done! Now you can connect to the router using several options: (user: root, password: root)
* telnet 192.168.31.1
* ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -c 3des-cbc -o UserKnownHostsFile=/dev/null root@192.168.31.1
* ftp: using a program like cyberduck
~~~

After runing the exploit script, I still cannot tenet to my Mi Router.
~~~shell
telnet 192.168.31.1
Trying 192.168.31.1...
telnet: connect to address 192.168.31.1: Connection refused
telnet: Unable to connect to remote host
~~~

primelyw, 2020-12-15:
firmware version is `miwifi_r4a_firmware_51508_2.28.38.bin`,whose md5 matches the one you show in README.md

primelyw, 2020-12-15:
exploit script version is the latest

aaronchin, 2020-12-15:
新购的4a firmware 已经变成2.28.65版了

acecilia, 2020-12-15:
I do have two xiaomi 4A gigabit routers and I can exploit without any issue. Most probably your network configuration is too complex. Try a simple network configuration or use exploit version `0.0.1`. Please read the readme and other issues in this repository to see how other people solved it

closed 2020-12-15 by acecilia

xvoidnessx, 2021-01-22:
Hey primelyw, 

I had your issue and managed to solve it. 

I am doing this in a vm, and it looked like that I must access the router 192.168.31.1 from within the VM to obtain the correct STOK. You can un comment 2 lines in the script to get print out status of the codes execution, the script is trying to upload a payload to the router and then execute it to start up the services, 

I would highly recommend we turn these on by default, 
remove the # infront of the 2 lines as below to get a more verbose output. 

 Line 69: print(r1.text)
 Line 78: print(r2.text)

Wrong STOK payload didnt get executed hence cant telnet 

Router IP address [press enter for using the default 192.168.31.1]: 192.168.31.1
stok: b041deacfad1569c57cab738d27b7135
****************
router_ip_address: 192.168.31.1
stok: b041deacfad1569c57cab738d27b7135
****************
start uploading config file...
{"code":401,"msg":"Invalid token"}
start exec command...
{"code":401,"msg":"Invalid token"}
done! Now you can connect to the router using several options: (user: root, password: root)
* telnet 192.168.31.1
* ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -c 3des-cbc -o UserKnownHostsFile=/dev/null root@192.168.31.1

this one is correct stok, you can see a different message, and the services are open for connections

Router IP address [press enter for using the default 192.168.31.1]: 192.168.31.1
stok: 2068901f541fd8255afabec832ea4d76
****************
router_ip_address: 192.168.31.1
stok: 2068901f541fd8255afabec832ea4d76
****************
start uploading config file...
{"code":1629,"msg":"解压失败，可能文件已经损坏"}
start exec command...
{"download":72.33,"bandwidth":0.57,"code":0}
done! Now you can connect to the router using several options: (user: root, password: root)
* telnet 192.168.31.1
* ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -c 3des-cbc -o UserKnownHostsFile=/dev/null root@192.168.31.1
* ftp: using a program like cyberduck