acecilia/OpenWRTInvasion #54: Unzip error

ravikumargrk, 2020-12-20:
```
ravi@DESKTOP-ODEG9VL:/mnt/d/router4C/OpenWRTInvasion$ python3 remote_command_execution_vulnerability.py
Router IP address [press enter for using the default 192.168.31.1]: 192.168.1.25
stok: fd74670fe216b01b93afe6190182ec43
****************
router_ip_address: 192.168.1.25
stok: fd74670fe216b01b93afe6190182ec43
****************
start uploading config file...
{"code":1629,"msg":"Unzip error, file is not intack"}
start exec command...
{"download":0,"bandwidth":0,"code":0}
done! Now you can connect to the router using several options: (user: root, password: root)
* telnet 192.168.1.25
* ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -c 3des-cbc -o UserKnownHostsFile=/dev/null root@192.168.1.25
* ftp: using a program like cyberduck

```
Product : Mi Router 4C 
Firmware :  2.14.128 [English] ( I can't find firmware that works : 2.14.81 )

acecilia, 2020-12-20:
Closing, as there is no explanation of the issue

closed 2020-12-20 by acecilia

ravikumargrk, 2020-12-20:
Hi, as you can see I get unzip error `Unzip error, file is not intack` and ssh  or telnet or ftp connections are refused. 
```
ravi@DESKTOP-ODEG9VL:/mnt/d/router4C/OpenWRTInvasion-0.0.1$ telnet 192.168.1.25
Trying 192.168.1.25...
telnet: Unable to connect to remote host: Connection refused
ravi@DESKTOP-ODEG9VL:/mnt/d/router4C/OpenWRTInvasion-0.0.1$ ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -c 3des-cbc -o UserKnownHostsFile=/dev/null root@192.168.1.25
ssh: connect to host 192.168.1.25 port 22: Connection refused
```

I cannot access ssh/telnet. Sorry I didn't explain it earlier.

acecilia, 2020-12-20:
I cant help much. Most of the issues reported here are due to complex network setups. Try using a simple network setup, or try using the other versions of the exploit that do not use internet. It is all explained in the readme 🙂

ravikumargrk, 2020-12-20:
Ok. I have gone through all the issues and did try first release (no internet needed). Same result.
I will now try using simple network setup. 
Thanks for fast response.

ravikumargrk, 2020-12-20:
Ok. I have gone through all the issues and did try first release (no internet needed). Same result.
I will now try using simple network setup.

By simple network setup, what do you mean? I have connected my PC to router via it's own Wifi. you want me to use Lan port instead

acecilia, 2020-12-20:
Make sure that the router has access to the internet when running the exploit, and that the IP of the router is the correct one. Try to avoid complex network configurations, or using raspberry pi, or some other weird setup

stpoa, 2022-09-04:
I have the same problem but whit 3c

EvgeniyBlinov, 2023-10-13:
> I have the same problem but whit 3c

+1