acecilia/OpenWRTInvasion #59: Exploit fixed шт 2.28.65

DigikaZ, 2021-01-13:
Hey, I can confirm this was fixed in very recent (2021 purchase) versions of the 4AGiga, the API to run netspeeed test no longer executes incorrect URL as shell commands and just returns default zero values (it did the same before but also would execute commands).

Add a note to main page, cheers!

acecilia, 2021-01-13:
I recently bought a 4A gigabit and was able to update to the latest available firmware `2.28.132` and exploit it without issues. Could you explain in more detail what do you mean?

acecilia, 2021-01-13:
Or provide any links?

acecilia, 2021-01-13:
Never hear about a `2.28.65` firmware version for the gigabit 4A, only `2.28.62` and `2.28.132`. Could you please add screenshots?

DigikaZ, 2021-01-14:
Neither did I! I swapped it for INT version, so sorry, cant provide detailed info anumore (int version costed me little extra, but it worked first time like a magic).

acecilia, 2021-01-17:
Ok, thanks for reporting. Closing this for now, I’ll wait and see if more reports appear about this new version before updating the readme

closed 2021-01-17 by acecilia

MeowCold, 2021-02-07:
Hi
This is a Chinese version, and the firmware is not officially provided. Temporarily only found in the routers newly purchased in China on 20.10.