acecilia/OpenWRTInvasion #61: R3L cannot working

LiarOnce, 2021-01-16:
Firmware is `2.9.217` but I also tried `2.14.45`.
0.0.1:
The `netcat` has no output.
[image]
0.0.6:
Error in telnet connection: Connection refused.
[image]

alberto0o, 2021-01-16:
Same problem with xiomi mi 3C, i have been looking and maybe some url is not working inside the script, maybe the urls from speedtest... I dont know hehe didnt check much, but i have same issue as you

LiarOnce, 2021-01-17:
I got an error: `{"code":1629,"msg":"unzip error, file is not intack"}` 
And it has nothing to do with the network environment:
[image]

LiarOnce, 2021-01-17:
Now it's working https://github.com/acecilia/OpenWRTInvasion/issues/62#issuecomment-761727533

closed 2021-01-17 by LiarOnce