acecilia/OpenWRTInvasion #68: Slack Link is outdated

krumelmonster, 2021-02-23:
The Slack link in the readme (https://join.slack.com/t/openwrt-workspace/shared_invite/zt-cz2m5uf4-Q8wbP_LKggOy9B7IQyaqfA) seems to be outdated:

> To join this workspace, you’ll need to ask the person who originally invited you for a new link.

I thought it would have been a convenient way to report that I successfully used the exploit on a Xiaomi 3Gv1 but I don't know the original firmware version. I had to set up the router with the chinese web interface (set it to AP mode, set SSID and PSK so I can access the router main menu) before it worked.

acecilia, 2021-02-23:
Thanks for reporting! Probably the slack is dead. I updated the readme mentioning the support for `Xiaomi 3Gv1` and removing the slack link. Thanks so much! 🙏

closed 2021-02-23 by acecilia

krumelmonster, 2021-03-23:
An unrelated note: I wrote a tool to automate OpenWRT on xiaomi router which is based on OpeWRTInvasion (but I plan to implement support for the AC2100 through another exploit soon)

https://github.com/krumelmonster/autowrt
https://forum.openwrt.org/t/autowrt-a-handsfree-installation-method-on-xiaomi-routers/91216

115606666, 2024-09-12:
Success report.
OS: Windows 10 Pro
Python: WinPython 3.12.2 (with venv)
Firmware: 2.28.14 (miwifi_r3g_firmware_a0d39_2.28.14.bin)
Note that the real-time protection of virus and threat needs to turn off otherwise the file busybox-mipsel under the script_tools will be quarantined and makes invasion fail.