acecilia/OpenWRTInvasion #69: won't work on 3.0.23 / 2.14.87; HTTP body attached

AddaxSoft, 2021-03-02:
Hello, 

I initially tested on factory firmware: `2.14.87` and didn't work, so I tried another firmware
from openWrt page: https://openwrt.org/toh/xiaomi/xiaomi_mi_router_4c
Flashed it into the router (4C) - firmware v `3.0.23`

It flashed fine and working correctly (the firmware), but then I tried the exploit
I followed the instructions with 0.0.6, and 0.0.1 but can't get it to execute the remote code.
telnet, ssh, ftp all are closed when I nmap the router.

Here is the response when the payload is sent:
(captured using wireshark)

```
POST /cgi-bin/luci/;stok=1ae5af04995d75dcb8e6fa853c832bf2/api/misystem/c_upload HTTP/1.1
Host: 192.168.31.1
User-Agent: python-requests/2.18.4
Accept-Encoding: gzip, deflate
Accept: */*
Connection: keep-alive
Content-Length: 1905
Content-Type: multipart/form-data; boundary=9f198d4413f044fd921cda49a87f01bd

--9f198d4413f044fd921cda49a87f01bd
Content-Disposition: form-data; name="image"; filename="payload.tar.gz"

......>`..payload.tar..W.n.6.....w8...FJQ...qb'u|A.&..vvQ...#qF.H.BR.q.?Y........i.KS...[....D....s..8.
.fT.T..,]... 
.C..t.w.....[A..=...
..p.....k....0..E.l;.7......s.u..J...{..x..}..SR...j/J..`.....].jO...Bf.K.).P...x/r....dF5.p.kt.......dg{{...[...^<.....G......13..dw.;......=.j..*...*....p....~....7@c~A.2M.nl....
.#%
..+_...f..n>..a.....].>o...E(..m....o..3..r=rwr.{s~.p..Iy_^S.....a.....3.......5n.(.7/a.....j.o.._..;...|..p.3.w.;.N...0..n...h..:.9e:..4V..K	.(.....[...M...$d).Q...K........E...5#.........7......	...y].D>.zW......ux.....= .e...[`X...1._...X.......L..P..\...4.....KmJ#.`.-d.?..1E
$.y. ..).....	* ...<.<...".<..?.....*3c..+...k.7.iJ...T1..D........2....%3`P(~!d.1.<*
..NN1U...tT.`j..l..	.*y..E..9.Ye@.........d)..TPj......)..	.T.......G"...)x........_.Th]rM...{....!....m.;.,.....x.D.bs.1m.Zv...2.t...D.y..,.GY..4....L..h.,.....\R;hi.DO.&...c...g..w.!.f.c....as...$...`.B_.@..!....C.w2...(<{V.....M.....!<\.Y..NS[..S.:..x.....)gy......3u........o.Qn".....z..........V..m.+n....*p.VB...<.kugN.k........!.f.P.#....H.bq1y..4.....h...RU`.........1...n..e9.c1N....R.r..,......D.8~b..e-)n\y
.[...z.y@k..P..Y!k...Dj3...?....>....K_.,....{l.
..g..n..2........u	^a.n.....%.+.......8..?..A:..c.L.?..d"8Ix.b.!S%."$.a._...b.I4.\..........U<.e.J....V.@K.(L...v...M....RL...{.._.9...E.y...	'.S%.......&...n..pR..MfL~]..5..u.e..6W.!.'xM.%6~..).#.S.....\.......z...h..U....j..zl.v......}........?.{;r..>V.Wtu..z.........W.w..f..w
h.P.w.]#Q.J..w=.nn.........*...BD..\........s..&B!..Qli}.c/.T..).....X...=.....U.78
nVXaf%.K....?...<.)w.z
.t..^........`.y......W?^...........5....>.M~.k.O..@....rkjY....+......M.9..NL........9.......%.G.W.bD.{h.......
.sk...c...,..	;.T...@..(...bMWke......{.u...$.vMR9_..p.n..wU.{....QG.u.QG.u.QG.u.QG.......n9A.(..
--9f198d4413f044fd921cda49a87f01bd--

--------------------------------------------------------------------

HTTP/1.1 200 OK
Server: nginx
Date: Tue, 02 Mar 2021 22:27:46 GMT
Content-Type: text/html; charset=utf-8
Content-Length: 61
Connection: close
Cache-Control: no-cache
Expires: Thu, 01 Jan 1970 00:00:01 GMT
MiCGI-Switch: 1 1
MiCGI-Client-Ip: 192.168.31.142
MiCGI-Host: 192.168.31.1
MiCGI-Http-Host: 192.168.31.1
MiCGI-Server-Ip: 192.168.31.1
MiCGI-Server-Port: 80
MiCGI-Status: CGI
MiCGI-Preload: no

{"code":1629,"msg":"......................................."}
```

I'm happy to troubleshoot with anyone.

WNinja, 2021-03-03:
It seems I have the same problems like you with  4A Gigabit，I change the code of remote_command_execution_vulnerability.py and find “print(r1.text)” will show {"code":1629,"msg":"解压失败，可能文件已经损坏"}， (the Chinese words means unzip fail and this file maybe broken)， is this router's bug has been fixed with version  2.28.62？

AddaxSoft, 2021-03-03:
@WNinja Thanks for your reply it helped me understand the problem and resolved the issue.

**The problem**: inability of Windows to properly terminate .tar files, doesn't matter if you use bash or python under windows it won't resolve the issue.

**the solution**: use Linux!

even though I get the error: 
[image]

I still get a shell back:
[image]

Here are the two files in Hex editor:
* top one is is the working one
* bottom one is the windows created one and is invalid / doesn't trigger the vuln
[image]

I hope this helps anyone struggling with the same issue!

acecilia, 2021-03-04:
@AddaxSoft thanks for troubleshooting! 🙏

closed 2021-03-04 by acecilia

WNinja, 2021-03-04:
I use Ubuntu 20.04.2 find it even doesn't work

AddaxSoft, 2021-03-19:
investigate why, maybe you have IP addresses wrong