acecilia/OpenWRTInvasion #70: Got shell but readonly filesystem? MiWiFi-R4A-2.28.65

springrider, 2021-03-05:
just bough brand new Xiaomi 4A Gigabit(Linux XiaoQiang 3.10.14 #1 MiWiFi-R4A-2.28.65 SMP Wed Mar 11 06:35:56 UTC 2020 mips GNU/Linux)

I tried v0.0.6, and v0.0.4, all the same:

```
Router IP address [press enter for using the default 192.168.31.1]:
stok: d154f331905f2fd45127f86f13babf9a
****************
router_ip_address: 192.168.31.1
stok: d154f331905f2fd45127f86f13babf9a
****************
start uploading config file...
{"code":1629,"msg":"解压失败，可能文件已经损坏"}
start exec command...
{"download":17.69,"bandwidth":0.14,"code":0}
done! Now you can connect to the router using several options: (user: root, password: root)
* telnet 192.168.31.1
* ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -c 3des-cbc -o UserKnownHostsFile=/dev/null root@192.168.31.1
* ftp: using a program like cyberduck
```
after this I can not telnet. 

tried v0.0.1 instead, and got 502 bad gayway error:
```

Start netcat on port 4444
(The way to do this in MacOS is to open a terminal and run '/usr/bin/nc -l 4444')
When you are done, press any key to continue
Router IP address: 192.168.31.1
Your IP address: 192.168.31.61
stok: d154f331905f2fd45127f86f13babf9a
****************
netcat_port: 4444
attacker_ip_address: 192.168.31.61
router_ip_address: 192.168.31.1
stok:d154f331905f2fd45127f86f13babf9a
****************
start uploading config file ...
{"code":1629,"msg":"解压失败，可能文件已经损坏"}
start exec command...
<html>
<head><title>502 Bad Gateway</title></head>
<body bgcolor="white">
<center><h1>502 Bad Gateway</h1></center>
<hr><center>nginx</center>
</body>
</html>

done!
```

on the /usr/bin/nc -l 4444 console, I actually got a shell, so the exploit works? but the filesystem is readonly, I can't do anything.  maybe that's the reason the scripts failed to execute?
```

 /usr/bin/nc -l 4444
/bin/sh: can't access tty; job control turned off
BusyBox v1.19.4 (2020-03-11 06:21:48 UTC) built-in shell (ash)
Enter 'help' for a list of built-in commands.

/www/cgi-bin # ls
luci
upload
/www/cgi-bin # touch test
touch: test: Read-only file system
/www/cgi-bin # cd ..
/www # ls
cgi-bin
cn
err
favicon.ico
img
index.html
init.html
js
luci-static
self_diag
self_diag.html
static
v3.html
vas
webinitrdr.html
xiaoqiang
/www # touch test
touch: test: Read-only file system

```

springrider, 2021-03-06:
sorry, didn't read readme carefully, maybe I need to "cd /tmp" as mentioned below:
If after reading above text you still want to proceed, after login to the router through telnet run the following commands:

```
cd /tmp
curl https://raw.githubusercontent.com/acecilia/OpenWRTInvasion/master/firmwares/OpenWrt/06-06-2020/openwrt-ramips-mt7621-xiaomi_mir3g-v2-squashfs-sysupgrade.bin --output firmware.bin # Put here the URL you want to use to download the firmware
./busybox sha256sum firmware.bin # Verify the firmware checksum before flashing, very important to avoid bricking your device!
mtd -e OS1 -r write firmware.bin OS1 # Install OpenWrt
This will install the snapshot version of OpenWrt (without Luci). You can now use ssh to connect to the router (and install Luci if you prefer it).
```

will try it later and see if it works.

acecilia, 2021-03-06:
I own 2 Xiaomi gigabit 4A and everything works perfectly :)

closed 2021-03-06 by acecilia

springrider, 2021-03-06:
thanks,  mine works too :),  using v0.0.6,  turns out the script.sh needs to download:
dropbearStaticMipsel.tar.bz2 and busybox-mipsel, but for the router the urls in script.sh is unreachable... so I download these two files and start a server locally, make the script.sh to download file from my mac, then everything works like a charm.

WNinja, 2021-03-08:
> thanks, just figured out the reason :), using v0.0.6, turns out the script.sh needs to download:
> dropbearStaticMipsel.tar.bz2 and busybox-mipsel, but for the router the urls in script.sh is unreachable... so I download these two files and start a server locally, make the script.sh to download file from my mac, then everything works like a charm.

I finally made it with your method, thanks !!

cibimo, 2021-04-19:
> thanks, just figured out the reason :), using v0.0.6, turns out the script.sh needs to download:
> dropbearStaticMipsel.tar.bz2 and busybox-mipsel, but for the router the urls in script.sh is [...]

It works!

zengyongxing, 2021-04-22:
> thanks, just figured out the reason :), using v0.0.6, turns out the script.sh needs to download:
> dropbearStaticMipsel.tar.bz2 and busybox-mipsel, but for the router the urls in script.sh is [...]

Really appreciated, it works.

t2krew, 2021-07-10:
> thanks, just figured out the reason :), using v0.0.6, turns out the script.sh needs to download:
> dropbearStaticMipsel.tar.bz2 and busybox-mipsel, but for the router the urls in script.sh is [...]

is it this solution for '{"code":1629,"msg":"解压失败，可能文件已经损坏"}' ?