acecilia/OpenWRTInvasion #76: Exploit not executed

onlinefabster, 2021-04-11:
Hello,

I just bought an Xiaomi MI Router 4A gigabit version which has firmware version 2.28.62. 
I did a basic setup and tried to run the exploit but without success. 
I see that the upload works alright, I even can add a test file in the tar file that ends in tmp directory.
[image]
[image]
[image]

But for some reason the busybox telnet deamon is not started, I cannot find it in the process list. 
I cannot connect to it, connection is refused. 
[xiaoqiang.log](https://github.com/acecilia/OpenWRTInvasion/files/6292508/xiaoqiang.log)

Any idea what could go wrong? 

Kind regards,
  Fabian

acecilia, 2021-04-13:
I cant help much, just to say that I exploited two `Xiaomi MI Router 4A gigabit` with the same version you mention without issues. Try to simplify your network setup, that is most of the times the issue

acecilia, 2021-04-13:
https://github.com/acecilia/OpenWRTInvasion/issues/75#issuecomment-816559455

closed 2021-04-13 by acecilia