acecilia/OpenWRTInvasion #79: OpenSSH is using Legacy Options

TaylorTWBrown, 2021-04-19:
Connecting with SSH requires the use of **diffie-hellman-group1-sha1**.

This is a legacy SSH option. Using this legacy feature has a practical drawback; Microsoft's implementation of OpenSSH on Windows does not include some legacy options, meaning that the script cannot be used on Windows.

Version 0.01 of the script doesn't rely on SSH, so it will probably work on Windows out of the box. However, newer versions will need to be run on a Linux host, or users will need to download other OpenSSH binaries to connect from Windows.

I am opening this issue as I hope to have some time to contribute code soon. I'd like it to be easier to use from Windows, as well as Linux and macOS. I like the project; thanks for making this!

acecilia, 2021-04-20:
You can use telnet 😄

acecilia, 2021-08-15:
Closing due to inactivity

closed 2021-08-15 by acecilia