acecilia/OpenWRTInvasion #80: Connection refused

dtlics, 2021-04-22:
I'm getting "telnet: Unable to connect to remote host: Connection refused"...

I tried to retype the command and reboot my router, none of them works.

I'm using Xiaomi Gigabyte Edition and it's connected to the internet and it's connected to my PC using an ethernet cable.

datanizing, 2021-04-23:
Same for me. I am using the router in access point mode BTW. Also tried updating to latest firmware 3.0.24, but no change.

datanizing, 2021-04-23:
Additional info: Uploading the tar.gz works fine, but then there is an error in `print(r1.text):

`{"code":1629,"msg":"Couldn't unzip, the file is corrupt"}`

acecilia, 2021-04-23:
Most likely you are having issues with your network setup. I cant help much there. There are multiple solved issues reporting the same thing in this repository, please take a look

closed 2021-04-23 by acecilia