acecilia/OpenWRTInvasion #82: Couldn't unzip, the file is corrupt

austonpramodh, 2021-05-05:
error:
```
start uploading config file...
{"code":1629,"msg":"Couldn't unzip, the file is corrupt"}
```

Firmware: MiWiFi Stable 3.0.16
I tried the exploit script v0.0.1 and v0.0.6. Both have same errors.

I am executing these scripts on WSL2.

V0.0.1
[image]

V0.0.6
[image]

acecilia, 2021-05-05:
Cant help much. This issue is usually due to a complex network configuration: try to simplify it. You can see other similar reported issues in this repository.

closed 2021-05-05 by acecilia

austonpramodh, 2021-05-05:
Yes, I understand now. I was able to successfully install openwrt. Thank you.

The issue is caused because the port 4444 exposed on PC might not be contactable from the router.

For the users experiencing the same issue:
When you start netcat, your pc starts listening at 4444. Then the exploit prepares a tar file with script and uploads it to the router. When the router executes the script, it tries connecting to the PC with the IP address provided to the script. 

In my case, maybe the router wasn't able to connect to the netcat session. So I ran same thing on raspberry pi and it worked. May be because it didn't have any firewall configured.  Also, you might wonder the stok doesn't work as it depends on the clients IP. I had SOCKS port forwarded ([how-to-setup-ssh-socks-tunnel-for-private-browsing](https://linuxize.com/post/how-to-setup-ssh-socks-tunnel-for-private-browsing/)) from raspberry pi to firefox, logged into the router and got the stok.

PR #100 (2021-10-02): Windows fixes
The main issue arises from the fact that Windows use CRLF (`\r\n`) as line ending and Unix-like OS's use just LF (`\n`). So, when we save processed `build/speedtest_urls.xml` file all original LF replaced with CRLF. Later on this unwanted excess CR characters can't be properly proceed by the router OS (custom Linux) and exploiting process is falling apart.
Fixes #82, #69 and probably others.

Also remove unused dependency (or am I missing something?).