acecilia/OpenWRTInvasion #87: Mi Router 4  requests  error

aiifan, 2021-07-11:
I tried various version
[image]

telnet: Unable to connect to remote host: Connection refused

`****************
start uploading config file...
200 {'code': 401, 'msg': 'Invalid token'}
start exec command...
200 {'code': 401, 'msg': 'Invalid token'}
done! Now you can connect to the router using several options: (user: root, password: root)
* telnet 192.168.31.1
* ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -c 3des-cbc -o UserKnownHostsFile=/dev/null root@192.168.31.1
* ftp: using a program like cyberduck`

acecilia, 2021-07-12:
You probably have a problem in your network setup. There are multiple github issues with the same problem that were resolved in the end, please take a look at them. For example: https://github.com/acecilia/OpenWRTInvasion/issues/53

closed 2021-07-12 by acecilia

wangzhenjjcn, 2021-09-29:
> 您的网络设置可能有问题。同一个问题的github问题有多个最终解决了，请看一下。例如：#53

你在53link了87，在87link了53，都是说网络问题，但是不知道是说什么网络问题。
代码里面是不是可以把err抛出认证下是否成功之后给出访问链接，好多人给issue应该是通病了。有时间研究下呗。