acecilia/OpenWRTInvasion #92: Confirm success installing Mi Router 4A 100M (R4AC)

ramone01, 2021-08-23:
Hello, i tried many times with fail, but i finally made it.
1-) You need that firmware  2.18.28 (https://bigota.miwifi.com/xiaoqiang/rom/r4ac/miwifi_r4ac_all_c4b35_2.18.28.bin)
2-) If device is bricked, use that zip to debricking while downgrade firmware.  mir4ac.zip (http://www.mediafire.com/file/45dc6jhpezw290e/mir4ac.zip/file?) its tftp methode to debrick router.
3-) Than you update version to 2.18.58, you just push bin file with web interface. (http://bigota.miwifi.com/xiaoqiang/rom/r4ac/miwifi_r4ac_firmware_e9eec_2.18.58.bin)

4-)After 2.18.58 with Chinese language rom, than boot debian 10 with vmware (includes gnome desktop). And start scripting. Just same as youtube video guide. (https://www.youtube.com/watch?v=VxzEvdDWU_s&t=143s)

Important note, The address you connect to via telnet, while script finishes, and the address you log in MUST BE the same.

i figured out with that methode. 

down to  miwifi_r4ac_all_c4b35_2.18.28.bin
update to miwifi_r4ac_firmware_e9eec_2.18.58.bin
debian 10 with gnome for run script and login web interface same vmware host.

i tried all versions of invasion, no problem all is OK.

vietchovui, 2021-09-24:
Failed to install on MiWiFi r4ac firmware version 2.18.58 with this openwrt firmware https://raw.githubusercontent.com/acecilia/OpenWRTInvasion/master/firmwares/OpenWrt/06-06-2020/openwrt-ramips-mt7621-xiaomi_mir3g-v2-squashfs-sysupgrade.bin
And now I'm unable to debrick it using MIWIFI tool or the Debrick tool provided by Hoddys.

Gisi0, 2021-12-05:
works great
router starts to blink after flashing, have to reboot and than it works

marufmks, 2022-04-06:
> Hello, i tried many times with fail, but i finally made it. 1-) You need that firmware 2.18.28 (https://bigota.miwifi.com/xiaoqiang/rom/r4ac/miwifi_r4ac_all_c4b35_2.18.28.bin) 2-) If device is [...]

i've followed Hoodys  method  and i got stuck here. "Download the openwrt-sysupgrade-image to /tmp and write it to the flash with mtd -r write <image> OS1 ". i have a downloaded openwrt firmware .how can i put it into /tmp and write the firmware? please help.

z4kio, 2022-05-20:
Worked for me. 
Thanks a lot!!!

ahmeedslom, 2022-07-27:
> Hello, i tried many times with fail, but i finally made it. 1-) You need that firmware 2.18.28 (https://bigota.miwifi.com/xiaoqiang/rom/r4ac/miwifi_r4ac_all_c4b35_2.18.28.bin) 2-) If device is [...]

can u share the openwrt_fmware.bin  for R4AC     or download link

ramone01, 2022-07-28:
Hello, im woking for now. I'll find it in my hard drive  and send it when I
get home.

27 Tem 2022 Çar 15:24 tarihinde ahmeedslom ***@***.***> şunu
yazdı:

devnoname120, 2022-08-15:
@ramone01 Friendly reminder :)

saifeealii, 2022-11-05:
> Hello, i tried many times with fail, but i finally made it. 1-) You need that firmware 2.18.28 (https://bigota.miwifi.com/xiaoqiang/rom/r4ac/miwifi_r4ac_all_c4b35_2.18.28.bin) 2-) If device is [...]

Thanks. Finally got this 2.18.58 on my router. but it's still not working with docker method. I am still getting this error:
Warning: the process has finished, but seems like ssh connection to the router is not working as expected.
* Maybe your firmware version is not supported, please have a look at ........

should i must load debian on vmware and do it with that method?

saifeealii, 2022-11-05:
Thanks. Finally got this 2.18.58 on my router. but it's still not working with docker method. I am still getting this error:
Warning: the process has finished, but seems like ssh connection to the router is not working as expected.
* Maybe your firmware version is not supported, please have a look at ........

should i must load debian on vmware and do it with that method?

saifeealii, 2022-11-05:
Alright, so let me add this here. on this version: 2.18.58, it wasn't working (openwrtinvasion couldn't connect to the router) but i downgraded the version to 2.18.51 and boom it worked now. you can get all Chinese versions from here: 
https://mirom.ezbox.idv.tw/en/miwifi/R4AC/roms-stable/

parbitwoas, 2024-03-09:
hello i  have Xiaomi Mi Router 4A (R4AC) and i cant use ssh or telnet to continue the process
can any one help me to fix it ?

PS C:\WINDOWS\system32> docker build -t openwrtinvasion https://github.com/acecilia/OpenWRTInvasion.git
[+] Building 3.8s (8/8) FINISHED                                                         docker:default
 => CACHED [internal] load git source https://github.com/acecilia/OpenWRTInvasion.git              1.9s
 => [internal] load metadata for docker.io/library/python:3-alpine                                 1.7s
 => [1/5] FROM docker.io/library/python:3-alpine@sha256:1a0501213b470de000d8432b3caab9d8de5489e94  0.0s
 => CACHED [2/5] WORKDIR /app                                                                      0.0s
 => CACHED [3/5] COPY requirements.txt ./                                                          0.0s
 => CACHED [4/5] RUN pip install -r requirements.txt                                               0.0s
 => CACHED [5/5] COPY . ./                                                                         0.0s
 => exporting to image                                                                             0.0s
 => => exporting layers                                                                            0.0s
 => => writing image sha256:05916f17e71f0f1a96b5a2c497951fe1333b554da1c2c90827db44ad170c06f8       0.0s
 => => naming to docker.io/library/openwrtinvasion                                                 0.0s

View build details: docker-desktop://dashboard/build/default/default/3ti2n9impklsbdam38zpgquza

What's Next?
  View a summary of image vulnerabilities and recommendations → docker scout quickview
PS C:\WINDOWS\system32> docker run --network host -it openwrtinvasion
Router IP address [press enter for using the default 'miwifi.com']: 192.168.31.1
Enter router admin password: pars@5546@
There two options to provide the files needed for invasion:
   1. Use a local TCP file server runing on random port to provide files in local directory `script_tools`.
   2. Download needed files from remote github repository. (choose this option only if github is accessable inside router device.)
Which option do you prefer? (default: 1)
****************
router_ip_address: 192.168.31.1
stok: 79d3f70d3f345727da02c655278dfd76
file provider: local file server
****************
start uploading config file...
start exec command...
local file server is runing on 0.0.0.0:47807. root='script_tools'
done! Now you can connect to the router using several options: (user: root, password: root)
* telnet 192.168.31.1
* ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -oHostKeyAlgorithms=+ssh-rsa -c 3des-cbc -o UserKnownHostsFile=/dev/null root@192.168.31.1
* ftp: using a program like cyberduck
PS C:\WINDOWS\system32> ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -oHostKeyAlgorithms=+ssh-rsa -c 3des-cbc -o UserKnownHostsFile=/dev/null root@192.168.31.1
kex_exchange_identification: Connection closed by remote host
Connection closed by 192.168.31.1 port 22

Eretick, 2024-03-31:
> it worked now

I tried .51 from your link but it didn't work for me. I tried both ocker vesion and linux (dual boot), still the same.

prash723, 2024-07-02:
I am trying with version 2.18.51 in Fedora with Virtual box no working can anyone help?

I am getting error _Maybe your firmware version is not supported, please have a look at ....._

legedug, 2024-09-13:
Confirm success installin openwrt from debricking, using version 2.18.58 to version 23.05.4 in Kali Linux. Use version r4a (not international version).
In kali linux open firewall port 22 and 21 than exploitation occur.

alcatraz-zz, 2025-03-17:
I also want to report that after A LOT of messing around I finally got it working.

I have a device marked R4AC bought in Europe. I assumed it was international and proceeded but that was just a dead end.

The fastest way for installing OpenWRT 24.10 on this R4AC router for me was to perform a debrick (even if it works) with 

http://www.mediafire.com/file/45dc6jhpezw290e/mir4ac.zip/file

with the default test.bin file.

Then the next step is to flash miwifi_r4ac_firmware_e9eec_2.18.58.bin in the web interface. (the version is already 2.18.58 but you must reflash this file)

The next step is to use OpenWRTInvasion with option 2.

Then you login and flash the CHINESE non international xiaomi_mi-router-4a-100m-squashfs-sysupgrade.bin

Done! 2.4 and 5 GHz wifi confirmed working.