acecilia/OpenWRTInvasion #95: Recently bought 4a Gigabit seems to have a different MDT

DutchShadow, 2021-09-10:
Hey, I run into the issue that the mdt doesn't find OS1

My mdt looks as follows:

`dev:    size   erasesize  name
mtd0: 07f80000 00020000 "ALL"
mtd1: 00080000 00020000 "Bootloader"
mtd2: 00040000 00020000 "Config"
mtd3: 00040000 00020000 "Bdata"
mtd4: 00040000 00020000 "Factory"
mtd5: 00040000 00020000 "crash"
mtd6: 00040000 00020000 "crash_syslog"
mtd7: 00040000 00020000 "cfg_bak"
mtd8: 00400000 00020000 "kernel0"
mtd9: 00400000 00020000 "kernel1"
mtd10: 01a00000 00020000 "rootfs0"
mtd11: 01a00000 00020000 "rootfs1"
mtd12: 02600000 00020000 "overlay"
mtd13: 01b80000 00020000 "obr"
mtd14: 00b43000 0001f000 "ubi_rootfs"
mtd15: 021e8000 0001f000 "data"
`

Any clue what's going on?

DutchShadow, 2021-09-10:
updated through stock UI to .175

Trying the Mi Router 4c telegram group method now

acecilia, 2021-09-13:
I have no idea what the issue can be

DutchShadow, 2021-09-15:
to which mtd should I try to write the sysupgrade.bin? I've managed to brick and unbrick already so I don't mind testing a few... 
mtd goes far over my head so idk if someone experienced with it wants to give me some tries...

acecilia, 2021-09-15:
Cant help here. Probably you are more successful writing in the openWrt forum: https://forum.openwrt.org/t/xiaomi-mi-router-4a-gigabit-edition-r4ag-r4a-gigabit-fully-supported-and-flashable-with-openwrtinvasion/36685

acecilia, 2021-09-24:
Lemme know if there are any updates, and I will reopen the issue

closed 2021-09-24 by acecilia