acecilia/OpenWRTInvasion #114: {"code":1629,"msg":"解压失败，可能文件已经损坏"} 最新版本报错 我把print打开

weilingwei, 2021-12-24:
stok: c4f1f0948134a8cd9de83ffed91b63a6
****************
start uploading config file...
http://192.168.31.1/cgi-bin/luci/;stok=c4f1f0948134a8cd9de83ffed91b63a6/api/misystem/c_upload
{"code":1629,"msg":"解压失败，可能文件已经损坏"}
start exec command...
请问这个怎么解决

Cxxxl, 2021-12-29:
同样的问题 小米4A 百兆版 固件降到2.18.51也不行 ;
payload.tar.gz文件本身是没有损坏的 可以正常解压 是我们刷的固件版本官方修复这漏洞了？
https://github.com/UltramanGaia/Xiaomi_Mi_WiFi_R3G_Vulnerability_POC/blob/master/report/report.md
这个报告里的远程任意文件读取漏洞在我本地测试也变成404了

Cxxxl, 2021-12-29:
翻了官方论坛升级公告 https://web.vip.miui.com/page/info/mio/mio/detail?postId=5365416&app_version=dev.20051
更新是在漏洞爆出后一个月出的2.18.51 日志里写的修复了安全漏洞 是修了上面报告里的漏洞么？
但是看readMe里 有用这个版本固件破解成功的 搞不明白了 求指点

acecilia, 2022-01-05:
Sorry, I do not understand

closed 2022-01-05 by acecilia

kivid, 2022-01-21:
小米R3A，固件2.18.40也存在这个问题，但反复测试偶尔能上传并解压成功，还没发现原因。

Xiaomi R3A, firmware 2.18.40 also has this problem. After repeated tests, it can be uploaded and decompressed successfully occasionally, but the reason has not been found.