acecilia/OpenWRTInvasion #125: Exploit working successfully on 4A Gigabit firmware ver. 3.10.18

diadatp, 2022-03-16:
Mi Router 4A Gigabit Edition

Model: R4A
Firmware Release 3.10.18
Manufactured in 07/2021
Flash Chip: Winbond W25Q128JVSIQ
Notes: Three shielding covers were missing from the PCB.

I used the docker image on an Ubuntu 18.04 host, connected via LAN cable.

For some reason using the default ("miwifi.com") as the router IP address resulted in an "Xiaomi router not found..." error. The program tried to continue anyway and even said that it was done.

```
Router IP address [press enter for using the default 'miwifi.com']: 
Xiaomi router not found...
You need to get the stok manually, then input the stok here: ■■■■■■■■
<< output omitted >>
Which option do you prefer? (default: 1)
****************
router_ip_address: miwifi.com
stok: ■■■■■■■■
file provider: local file server
****************
start uploading config file...
start exec command...
local file server is runing on 0.0.0.0:60451. root='script_tools'
done! Now you can connect to the router using several options: (user: root, password: root)
<< output omitted >>
```

However, using the IP address of the router directly, resulted in a successful exploit.

```
Router IP address [press enter for using the default 'miwifi.com']: 192.168.31.1
Enter router admin password: ■■■■■■■■
<< output omitted >>
Which option do you prefer? (default: 1)
****************
router_ip_address: 192.168.31.1
stok: ■■■■■■■■
file provider: local file server
****************
start uploading config file...
start exec command...
local file server is runing on 0.0.0.0:52791. root='script_tools'
local file server is getting 'busybox-mipsel' for 192.168.31.1.
local file server is getting 'dropbearStaticMipsel.tar.bz2' for 192.168.31.1.
done! Now you can connect to the router using several options: (user: root, password: root)
<< output omitted >>
```

Thank you to all the contributors for all the hard work!

chigarow, 2022-03-23:
hello, just wanna ask

i want to install [dnscrypt](https://github.com/DNSCrypt/dnscrypt-proxy) on my router, if i already exploit the router so can i do whatever i want? without changing the router firmware.

acecilia, 2022-03-24:
Mmm access to the router gives you the ability to do some things, but I would say that for installing packages you should install openWrt

chigarow, 2022-03-24:
ah i see, i thought i can install anything without openwrt by just exploiting the router
thankyou for your information!!