acecilia/OpenWRTInvasion #126: R4AC Brick Chinese Version

tuanha2000vn, 2022-03-19:
Hi, I've follow steps and the script Unlocking OS1 ...Erasing OS1 ...Writing from firmware.bin to OS1 ...Rebooting ...

Not sure what wrong but the router brick (yellow and blue light for hours)

What can I do to fix the issue? Thank you.

```
admin@Mac-mini-Late-2014 OpenWRTInvasion % python3 remote_command_execution_vulnerability.py
Router IP address [press enter for using the default 'miwifi.com']: 192.168.31.1
Enter router admin password: 11111111
There two options to provide the files needed for invasion:
   1. Use a local TCP file server runing on random port to provide files in local directory `script_tools`.
   2. Download needed files from remote github repository. (choose this option only if github is accessable inside router device.)
Which option do you prefer? (default: 1)2
****************
router_ip_address: 192.168.31.1
stok: a96b4f855ac9defcff760aae754f8901
file provider: remote github repository
****************
start uploading config file...
start exec command...
done! Now you can connect to the router using several options: (user: root, password: root)
* telnet 192.168.31.1
* ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -c 3des-cbc -o UserKnownHostsFile=/dev/null root@192.168.31.1
* ftp: using a program like cyberduck
admin@Mac-mini-Late-2014 OpenWRTInvasion % telnet 192.168.31.1
Trying 192.168.31.1...
Connected to 192.168.31.1.
Escape character is '^]'.

XiaoQiang login: root
Password: 

BusyBox v1.19.4 (2019-11-26 08:37:52 UTC) built-in shell (ash)
Enter 'help' for a list of built-in commands.

 -----------------------------------------------------
       Welcome to XiaoQiang!
 -----------------------------------------------------
  $$$$$$\  $$$$$$$\  $$$$$$$$\      $$\      $$\        $$$$$$\  $$\   $$\
 $$  __$$\ $$  __$$\ $$  _____|     $$ |     $$ |      $$  __$$\ $$ | $$  |
 $$ /  $$ |$$ |  $$ |$$ |           $$ |     $$ |      $$ /  $$ |$$ |$$  /
 $$$$$$$$ |$$$$$$$  |$$$$$\         $$ |     $$ |      $$ |  $$ |$$$$$  /
 $$  __$$ |$$  __$$< $$  __|        $$ |     $$ |      $$ |  $$ |$$  $$<
 $$ |  $$ |$$ |  $$ |$$ |           $$ |     $$ |      $$ |  $$ |$$ |\$$\
 $$ |  $$ |$$ |  $$ |$$$$$$$$\       $$$$$$$$$  |       $$$$$$  |$$ | \$$\
 \__|  \__|\__|  \__|\________|      \_________/        \______/ \__|  \__|

root@XiaoQiang:~# cd /tmp
root@XiaoQiang:/tmp# curl https://raw.githubusercontent.com/acecilia/OpenWRTInvasion/master/firmwares/OpenWrt/06-06-2020/openwrt-ramips-mt7621-xiaomi_mir3g-v2-squashfs-sysupgrade.
bin --output firmware.bin # Put here the URL you want to use to download the firmware
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 3840k  100 3840k    0     0  1960k      0  0:00:01  0:00:01 --:--:-- 1962k
root@XiaoQiang:/tmp# ./busybox sha256sum firmware.bin
83feed9ff633863acb1f14b61c30029b924fec252c1ed1e4de2a909e52b2d872  firmware.bin
root@XiaoQiang:/tmp# mtd -e OS1 -r write firmware.bin OS1
Unlocking OS1 ...
Erasing OS1 ...

Writing from firmware.bin to OS1 ...     
Rebooting ...
```

hajking, 2022-03-25:
[https://openwrt.org/toh/xiaomi/mi_router_4a_mir4a_100m?s[]=xiaomi&s[]=mi&s[]=router&s[]=4a&s[]=100m&s[]=edition#installation](url)

closed 2022-03-27 by acecilia

RadioOperator, 2022-03-28:
please check if your router used a new flash IC, likes my new xiaomi-4C, refer to:
https://forum.openwrt.org/t/new-xiaomi-4c-cannot-install-openwrt-flash-chip-changed-to-en25qx128a/123634