acecilia/OpenWRTInvasion #133: When running in docker on windows, wrong stok seems to be detected

schweini, 2022-04-21:
I can;t seem to get this to work - I run this following the windows/docker instructions. I don't get any errors, but can't connect via telnet, ssh or any other means after running this.

I did notice however that the stok shown in the script output differs completely from the stok seen in the browser URL of the UI.
I hace seen that this might be IP related, but I am running the commands on the same PC where I am running the docker commands.

Maybe some issue with the docker-ed script in combination with the seemingly recent change of auto-detecting the stok?

BrilliantStanley, 2022-05-06:
Same for me

marianomd, 2022-05-13:
Similar here, 4A gigabit, MiWiFi Release 3.0.24.

```
Xiaomi router not found...
You need to get the stok manually, then input the stok here:
```
I enter it manually, and after the process ends, no telnet or ssh port are open.

Tried with and without docker.

Maybe this version is patched?

marianomd, 2022-05-13:
My flash: GigaDevice GD25Q128E

Seems someone managed to flash OpenWRT using other tools on this version of the router. But no time yet to investigate that path: [](https://forum.openwrt.org/t/xiaomi-4a-3-gigaport-cannot-flash-openwrt-21-02/122878/17)

RadioOperator, 2022-05-14:
re-try it on a pure ubuntu PC.

marianomd, 2022-05-17:
I managed to flash OpenWRT and Padavan with this [solution](https://forum.openwrt.org/t/xiaomi-4a-3-gigaport-cannot-flash-openwrt-21-02/122878/18).

acecilia, 2022-09-01:
Closing due to inactivity. I would suggest to just pass the stok manually

closed 2022-09-01 by acecilia