acecilia/OpenWRTInvasion #136: new firmware 2.30.20 seems not work

NickeryXu, 2022-06-12:
Router: Mi 4A Gigabit edition
Firmware Version: 2.30.20
System: Windows 10
OpenWrtInvasion version: 0.0.8

no exception in scripts, but telnet failed; I choose local TCP, looks like build a local tcp server, so that shouldn't be a network problem; I have already enable telnet client; the only reason I can find is this firmware version might change api or something, realy needs help :(
[image]

devidlis123, 2022-06-19:
Hello, mine is the same version. No solution?

NickeryXu, 2022-06-19:
> Hello, mine is the same version. No solution?
Yup, finally I change my device to Redmi AX6S, got a snopshot OpenWRT and that works fine

borekon, 2022-06-21:
Duplicate from #135

acecilia, 2022-06-21:
Thanks @borekon 🙌 Closing as duplicate