acecilia/OpenWRTInvasion #140: May not work with firmware 3.2.26

arihid, 2022-08-17:
Log shown success, but I can't ssh/telnet/ftp to this.
Here is the log:
```C:\Users\<myuser>\Downloads\OpenWRTInvasion-0.0.8>python remote_command_execution_vulnerability.py
Router IP address [press enter for using the default 'miwifi.com']: 192.168.31.1
Enter router admin password: 12345678
There two options to provide the files needed for invasion:
   1. Use a local TCP file server runing on random port to provide files in local directory `script_tools`.
   2. Download needed files from remote github repository. (choose this option only if github is accessable inside router device.)
Which option do you prefer? (default: 1)1
****************
router_ip_address: 192.168.31.1
stok: <cleaned>
file provider: local file server
****************
start uploading config file...
start exec command...
local file server is runing on 0.0.0.0:61653. root='script_tools'
done! Now you can connect to the router using several options: (user: root, password: root)
* telnet 192.168.31.1
* ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -c 3des-cbc -o UserKnownHostsFile=/dev/null root@192.168.31.1
* ftp: using a program like cyberduck```

acecilia, 2022-09-08:
Version 3.0.10 is reported to work: https://github.com/acecilia/OpenWRTInvasion/issues/145

arihid, 2022-09-08:
I tried to downgrade, but it won't accept downgrade, unlike AX6S that accepts downgrade to internal beta.
I went to flash programmer route and it succeeded, but I will leave this open for future reference.

acecilia, 2022-09-08:
Did you manage to make it work then?

arihid, 2022-09-08:
Yes, it's running the current stable version of OpenWrt.

acecilia, 2022-09-08:
If you explain in detail how you did it, I can add it to the readme and close this issue (I do not know what "flash programmer" is)

acecilia, 2022-09-08:
Also, which operating system did you use?

arihid, 2022-09-09:
[This post](https://forum.openwrt.org/t/xiaomi-wifi-router-3g-v2/42584/46) here described how to read the firmware, edit the necessary value and flash it back using an spi flash programmer, as firstly discovered and described in [this post](https://forum.openwrt.org/t/xiaomi-mi-router-4a-gigabit-edition-r4ag-r4a-gigabit-fully-supported-and-flashable-with-openwrtinvasion/36685).
It is reported that reading the flash content in Windows tend to yield corrupted image, so I used Ubuntu 22 on VM and it worked.
The chip is supposed to "GD25Q127C/GD25Q128C", not "GD25Q128C" for this particular device.

ElclarkKuhu, 2022-09-09:
Anyone Have 3.2.26 Firmware File? I have access to the shell but I accidentally flash corrupted file. and I tried to flash with Global 3.0.24 and Chinese 2.28.62 both fail using either MIWIFIRepairTool or PXE Server.

acecilia, 2022-09-09:
Added a mention to the readme, thanks!

closed 2022-09-09 by acecilia

tangmingxing1988, 2022-09-17:
> 

Sorry to bother you, Should I use TTL to USB adapter to interact with UART?

This is not listed in the post but in the referenced post.

arihid, 2022-09-17:
> > 
> 
> Sorry to bother you, Should I use TTL to USB adapter to interact with UART?
> 
> This is not listed in the post but in the referenced post.

Yes, use USB to TTL to interact with UART.

> Anyone Have 3.2.26 Firmware File? I had access to the shell but I accidentally flash corrupted file. and I tried to flash with Global 3.0.24 and Chinese 2.28.62 both fail using either MIWIFIRepairTool or PXE Server.

I only have my dump, but I won't recommend using it.

arihid, 2022-10-11:
Yes, use USB to TTL to interact with UART. I have PL2303, the driver is quite bothersome to use under Win 11, but trouble-free under Ubuntu 22.

Best Regards,

Arief Hidayat

awannawa, 2022-12-07:
> 

Hello Sir do you still have backup stock frimware 3.2.26? Can i get it?

arihid, 2022-12-08:
For everyone who need 3.2.26 firmware dump. I will attach my dumped firmware. These following values has been purged, thus you will need to edit the values prior flashing. Flash this firmware with SPI flash programmer after adjusting the values accordingly.
**DO NOT FLASH THE FILE AS IS, YOUR DEVICE INFORMATION WILL BE OVERWRITTEN!!**

Current modified NVRAM values:

```
bootdelay=0
ethaddr="00:AA:BB:CC:DD:10"
SN=12345/A1RQ01234
wl0_ssid=Xiaomi_AABB_CCDD_5G
wl1_ssid=Xiaomi_AABB_CCDD
nv_wifi_pwd=12345678
CountryCode=ID
```

I left `normal_firmware_md5` and `nv_sys_pwd` as is. I guess `normal_firmware_md5` value would be required to verify the firmware, and `nv_sys_pwd` is encrypted password string upon setup (usually same as WiFi Password). WebUI password shall be 12345678. If you can't login to the WebUI, try resetting the router once.

@ElclarkKuhu sorry if this took a long time. Back then I decided not to share my dumped firmware for reasons. Idk either this might work or not on your device.

[r4a.zip](https://github.com/acecilia/OpenWRTInvasion/files/10181106/r4a.zip)

awannawa, 2022-12-08:
> For everyone who need 3.2.26 firmware dump. I will attach my dumped firmware. These following values has been purged, thus you will need to edit the values prior flashing. Flash this firmware with SPI flash programmer after adjusting the values accordingly. **DO NOT FLASH THE FILE AS IS, YOUR DEVICE INFORMATION WILL BE OVERWRITTEN!!**
> 
> Current modified NVRAM values:
> 
> ```
> bootdelay=0
> ethaddr="00:AA:BB:CC:DD:10"
> SN=12345/A1RQ01234
> wl0_ssid=Xiaomi_AABB_CCDD_5G
> wl1_ssid=Xiaomi_AABB_CCDD
> nv_wifi_pwd=12345678
> CountryCode=ID
> ```
> 
> `bootdelay` is timer to delay boot process in seconds. 0 means it will not wait for user interactions. Change this value to >0. I left `normal_firmware_md5` and `nv_sys_pwd` as is. I guess `normal_firmware_md5` value would be required to verify the firmware, and `nv_sys_pwd` is encrypted password string upon setup (usually same as WiFi Password). WebUI password shall be 12345678. If you can't login to the WebUI, try resetting the router once.
> 
> @ElclarkKuhu sorry if this took a long time. Back then I decided not to share my dumped firmware for reasons. Idk either this might work or not on your device.
> 
> [r4a.zip](https://github.com/acecilia/OpenWRTInvasion/files/10181106/r4a.zip)

thankyou for dumped fw 3.2.26 sir, can i just flash with this bin file with Tiny PXE Server 1.0.0.23?

arihid, 2022-12-08:
> thankyou for dumped fw 3.2.26 sir, can i just flash with this bin file with Tiny PXE Server 1.0.0.23?

I don't know, can't recommend

awannawa, 2022-12-08:
> > thankyou for dumped fw 3.2.26 sir, can i just flash with this bin file with Tiny PXE Server 1.0.0.23?
> 
> I don't know, can't recommend

thankyou sir for fw dump...
stil finding best solution without spi hehehe

yonathanarya, 2022-12-12:
> [This post](https://forum.openwrt.org/t/xiaomi-wifi-router-3g-v2/42584/46) here described how to read the firmware, edit the necessary value and flash it back using an spi flash programmer, as [...]

How did you manage to install OpenWRT on this firmware version? I got the same router, but still can't find how to install it. I already flash the SPI chip and can control UART, but I can't flash it from there, either. Did I miss something?

yonathanarya, 2022-12-12:
Forget it, i manage to install OpenWRT successfully now. Since I can access the UART, I also can flash the file from there by downloading the binary using wget and flash using mtd.

NaufalF22, 2024-09-11:
[image]
Just try myself and i can access telnet without flashing the SPI chip

kopijahe, 2025-03-13:
I confirm the `0.0.8` release worked with this firmware, but you have to use telnet and FTP to connect and transfer file,

rfrz, 2025-04-27:
can confirm it worked with 0.0.8.

Tried with RHEL 9.5 inside vmware workstation pro 17.

but somehow when i tried using opensuse tumbleweed that was running natively, it refuses to connect

NaufalF22, 2025-04-27:
> didn't work in mine using latest version
> 
> testing using version 0.0.8 and receiving success message but when i tried to telnet i'm getting this error
> 
> `Could not open connection to the host, on port 23: Connect failed `
> 
> ftp also unable to connect

Try using linux

kopijahe, 2025-04-27:
You can also try using a static IP in the lower range (like  between `192.168.31.2 - 192.168.31.10`).

In my experience when using the script in termux on android, using those IPs lead to more success.

NaufalF22, 2025-07-07:
Just install it