acecilia/OpenWRTInvasion #143: May not work with 3.0.23 of miwifi 4c

Kinuseka, 2022-09-04:
Shows successful process but can not connect to ssh/telnet/ftp
```
python3 remote_command_execution_vulnerability.py
Router IP address [press enter for using the default 'miwifi.com']:
Xiaomi router not found...
You need to get the stok manually, then input the stok here: eeeeb4238c47e03055929549ffdd527b
There two options to provide the files needed for invasion:
   1. Use a local TCP file server runing on random port to provide files in local directory `script_tools`.
   2. Download needed files from remote github repository. (choose this option only if github is accessable inside router device.)
Which option do you prefer? (default: 1)2
****************
router_ip_address: miwifi.com
stok: eeeeb4238c47e03055929549ffdd527b
file provider: remote github repository
****************
start uploading config file...
start exec command...
done! Now you can connect to the router using several options: (user: root, password: root)
* telnet miwifi.com
* ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -c 3des-cbc -o UserKnownHostsFile=/dev/null root@miwifi.com

ssh: connect to host miwifi.com port 22: Connection refused
```

Kinuseka, 2022-09-04:
### UPDATE: 
I downgraded my router (mi router 4c) to 2.14.75 and the exploit now works properly.

Link for the ROM: https://mirom.ezbox.idv.tw/en/miwifi/R4CM/roms-stable/

**Do note that this is a third party website acting as a mirror server. Do it at your own risk**

Kinuseka, 2022-09-04:
After further investigation and testing, turns out that the exploit only works properly on Linux OS's and seems to not work on windows (atleast on my environment)? 

I can now confirm in my case, that 3.0.23 does indeed work with OpenWRTInvasion and the malfunction may be just related to OS environment misconfiguration.

closed 2022-09-04 by Kinuseka

tangmingxing1988, 2022-09-16:
> ### UPDATE: 
> 
> I downgraded my router (mi router 4c) to 2.14.75 and the exploit now works properly.
> 
> 
> 
> Link for the ROM: https://mirom.ezbox.idv.tw/en/miwifi/R4CM/roms-stable/
> 
> 
> 
> **Do note that this is a third party website acting as a mirror server. Do it at your own risk**

May I ask how to downgrade? When I use the web ui, it can't pass the file check.

lydacious, 2025-01-07:
> After further investigation and testing, turns out that the exploit only works properly on Linux OS's and seems to not work on windows (atleast on my environment)?
>
> I can now confirm in my case, [...]

Hello, I am on Ubuntu and trying to figure out the problem. I tried with both 3.0.23 and 2.14.75. 
Both give me the same error:
`file provider: local file server
****************
start uploading config file...
start exec command...
local file server is runing on 0.0.0.0:57567. root='script_tools'
Warning: the process has finished, but seems like ssh connection to the router is not working as expected.
* Maybe your firmware version is not supported, please have a look at https://github.com/acecilia/OpenWRTInvasion/blob/master/README.md#unsupported-routers-and-firmware-versions
* Anyway you can try it with: telnet`

How did you make it work with 3.0.23?