acecilia/OpenWRTInvasion #145: [SUCCESS] Mi Router 4A International Edition 100M (R4AC) on firmware 3.0.10

ValerieOSS, 2022-09-08:
Hey there, this exploit seems to work on the latest firmware for R4AC (3.0.10), exploit v0.0.9
Tested under Arch Linux
No workaround needed, as expected Windows does not work

```[luna@ideapad OpenWRTInvasion-0.0.9]$ python remote_command_execution_vulnerability.py
Router IP address [press enter for using the default 'miwifi.com']: 192.168.31.1
Enter router admin password: ■■■■■■■■
There two options to provide the files needed for invasion:
   1. Use a local TCP file server runing on random port to provide files in local directory `script_tools`.
   2. Download needed files from remote github repository. (choose this option only if github is accessable inside router device.)
Which option do you prefer? (default: 1)
****************
router_ip_address: 192.168.31.1
stok: ■■■■■■■■
file provider: local file server
****************
start uploading config file...
start exec command...
local file server is runing on 0.0.0.0:34171. root='script_tools'
done! Now you can connect to the router using several options: (user: root, password: root)
* telnet 192.168.31.1
* ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -c 3des-cbc -o UserKnownHostsFile=/dev/null root@192.168.31.1
* ftp: using a program like cyberduck
````

[image]

acecilia, 2022-09-08:
Awesome, thanks for the report 🙏

acecilia, 2022-09-08:
Added a mention to the readme :)

closed 2022-09-08 by acecilia