acecilia/OpenWRTInvasion #150: [Success] Mi 4a Gigabit firmware 3.10.18

menubboi, 2022-09-29:
Indian unit, used docker for exploit. But ftp connection was rejected by the router.

acecilia, 2022-09-29:
Added to the readme, thanks!

closed 2022-09-29 by acecilia

firefoxOnFire, 2023-05-04:
> Indian unit, used docker for exploit. But ftp connection was rejected by the router, Used Docker solution in windows for this.

Hey from where did you buy? can i know the version before buying?

firefoxOnFire, 2023-05-17:
> @firefoxOnFire I can probably add to this, I purchased the Mi Router 4A Gigabit from [Flipkart](https://www.flipkart.com/mi-router-4a-gigabit-1200-mbps/p/itma801f2bdfae50) and received the unit yesterday. The manufacturing date printed on the box was 10/2021 and came with firmware `3.10.18` same as @menubboi.
> 
> I initially setup the device and tested if everything is working. I then ran the [exploit script](https://github.com/acecilia/OpenWRTInvasion#using-the-command-line) from a Ubuntu WSL terminal from Windows. The below is the script output.
> 
> ```
> varkey@mjolnir:~/OpenWRTInvasion$ python3 remote_command_execution_vulnerability.py
> Router IP address [press enter for using the default 'miwifi.com']: 192.168.31.1
> Enter router admin password: <password>
> There two options to provide the files needed for invasion:
>    1. Use a local TCP file server runing on random port to provide files in local directory `script_tools`.
>    2. Download needed files from remote github repository. (choose this option only if github is accessable inside router device.)
> Which option do you prefer? (default: 1)1
> ******
> router_ip_address: 192.168.31.1
> stok: <stok>
> file provider: local file server
> ******
> start uploading config file...
> start exec command...
> local file server is runing on 0.0.0.0:52081. root='script_tools'
> Warning: the process has finished, but seems like ssh connection to the router is not working as expected.
> * Maybe your firmware version is not supported, please have a look at https://github.com/acecilia/OpenWRTInvasion/blob/master/README.md#unsupported-routers-and-firmware-versions
> * Anyway you can try it with: telnet 192.168.31.1
> ```
> 
> I just had to provide the IP address and admin password. The `stok` was retrieved automatically. I chose the `use local TCP file server` option, but only later it occurred to me that the local server running on Ubuntu WSL may not be reachable from the Mi router. Which is probably why SSH didn't work.
> 
> Anyway, I was able to telnet into the router, however FTP did not work (similar to @menubboi). I ended up directly downloading the OpenWRT firmware using `wget`. Note that `HTTPS` is not supported so you need to use an `HTTP` link which doesn't auto redirect to `HTTPS`, I used one of the OpenWRT mirrors.
> 
> After that ran the command to write the firmware, which took a few minutes to complete and the device rebooted.
> 
> ```
> root@XiaoQiang:/tmp# wget http://mirror.0x.sg/openwrt/releases/22.03.5/targets/ramips/mt7621/openwrt-22.03.5-ramips-mt7621-xiaomi_mi-router-4a-gigabit-squashfs-sysupgrade.bin
> wget http://mirror.0x.sg/openwrt/releases/22.03.5/targets/r
> amips/mt7621/openwrt-22.03.5-ramips-mt7621-xiaomi_mi-router-4a-gigabit-squashfs-
> sysupgrade.bin
> Connecting to mirror.0x.sg (118.189.187.101:80)
> openwrt-22.03.5-rami 100% |*******************************|  6400k  0:00:00 ETA
> root@XiaoQiang:/tmp# ls -l openwrt.bin
> ls -l openwrt.bin
> -rw-r--r--    1 root     root       6554224 May 17 11:47 openwrt.bin
> root@XiaoQiang:/tmp# busybox sha256sum openwrt.bin
> busybox sha256sum openwrt.bin
> sha256sum: applet not found
> root@XiaoQiang:/tmp# md5sum openwrt.bin
> md5sum openwrt.bin
> 5c931d7c5dab8911da8416c5b142fbdf  openwrt.bin
> root@XiaoQiang:/tmp# mtd -e OS1 -r write openwrt.bin OS1
> mtd -e OS1 -r write openwrt.bin OS1
> Unlocking OS1 ...
> Erasing OS1 ...
> 
> Writing from openwrt.bin to OS1 ...
> Rebooting ...
> ```
> 
> The `busybox` command to check the `sha256sum` did not work, so I ended up verifying the `md5sum` as a last resort. This is also probably because I ran the script from WSL Ubuntu and nothing could be fetched from the local file server.

Finally booted to openwrt?? what is the space left after installing openwrt??

varkey, 2023-05-17:
@firefoxOnFire Yep, after that it booted into OpenWRT. Space left is 8MiB. 

[image]

firefoxOnFire, 2023-05-17:
> @firefoxOnFire Yep, after that it booted into OpenWRT. Space left is 8MiB.
> [image]

> Space left is 8MiB.

Thanks.

iqs99, 2023-10-14:
Can someone help me!  I'm using the same 3.10.18 firmware. I want to connect my router with WISP. I tried connecting it through Wireless repeater mode but DHCP server is disabled and their is no setting provided to enable DHCP server in the router, while using it in Wireless repeater mode. My WISP requires router's DHCP server should be set Enable to use the service. Let me know if someone have solution. Thanks

Zaratussstra, 2024-08-08:
It turned out to flash openwrt-23.05.4. First, I installed firmware 3.0.24 using TinyPXE, then flashed scripts with https://4pda.to/forum/index.php?showtopic=905966&view=findpost&p=95240419. 
It was not possible to download the firmware via telnet, as it says here.

SPEED-OX, 2025-07-15:
> [@firefoxOnFire](https://github.com/firefoxOnFire) I can probably add to this, I purchased the Mi Router 4A Gigabit from [...]

> [@firefoxOnFire](https://github.com/firefoxOnFire) I can probably add to this, I purchased the Mi Router 4A Gigabit from [...]

@firefoxOnFire I need confirmation if the unit I have is also supported, Mine has the manufacture date of 07/2021, I bought that from flipkart and it has the exact same hardware as this [image](https://openwrt.org/_detail/media/xiaomi/xiaomi-mi-router-4a-gigabit-board-without-shielding-small.jpg?id=inbox%3Atoh%3Axiaomi%3Axiaomi_mi_router_4a_gigabit_edition) with firmware `3.10.18` . So it's likely the R4A Gigabit Edition V1

So I guess you can help me to install OpenWRT