acecilia/OpenWRTInvasion #152: Xiaomi Mi Router R4AG. Can`t complete OpenWrt install successfully

azimut3, 2022-10-22:
For those who struggle with installing OpenWrt after the invasion script is applied, check if you selected the correct OpenWrt version to install on your router. I downloaded the stable release from this page as it was suggested in one of the comments on a [forum ](https://forum.openwrt.org/t/xiaomi-mi-router-4a-gigabit-edition-r4ag-r4a-gigabit-fully-supported-and-flashable-with-openwrtinvasion/36685/). The link for stable release [download page](https://downloads.openwrt.org/releases/19.07.2/targets/ramips/mt76x8/).
When I used a link for release from this page I was forced to use _curl --insecure link_ to run it and install it.

In case you bricked your device, you can follow the instructions in this [comment](https://forum.openwrt.org/t/xiaomi-mi-router-4a-gigabit-edition-r4ag-r4a-gigabit-fully-supported-and-flashable-with-openwrtinvasion/36685/747?page=37) 
The only difference for me was the fact that it worked only with automatic IP settings and sometimes it took a few reboots of the router before firmware was automatically uploaded by MIWIFIRepairTool. Just tried a few times before it worked.

Hopefully, this note can save someone`s time significantly.
Best of luck

RadioOperator, 2022-10-23:
more remarks before you try to install openwrt on R4A / R4AG:
1. If the rom IC is Eon or CFeon EN25QX128A, Do Not use any official openwrt build, because the kernel not support for now, you have to use a special build for the chip.
2. If your xiaomi factory fw is version 2.30.20 or later, be careful it's the R4AG V2, openwrt still not support the version.

RadioOperator, 2022-10-23:
@azimut3 did you try to upgrade the factory fw to 2.28.69 ?

I'm using 2.28.69 on my 4AG, no Bus Error on it, any openwrt build ok.

Veoxys0910, 2022-10-23:
@RadioOperator It was me that got the Bus Error, but I still was able to install OpenWRT.

azimut3, 2022-10-23:
> @azimut3 did you try to upgrade the factory fw to 2.28.69 ?
> 
> I'm using 2.28.69 on my 4AG, no Bus Error on it, any openwrt build ok.

I was able to install everything successfully and decided to give a hint to someone who can be stuck like me.
My Mi firmware version was 2.18.58. I also had a bus error and my issue was resolved with another version of OpenWrt that I left a link to in the original comment. 
Thanks for your attention. Maybe 2.28.69 can also be an option to resolve such an issues

Should I close this issue as it is resolved? Should I add any specific comments to it maybe?

closed 2022-10-23 by azimut3

RadioOperator, 2022-10-23:
@azimut3  Sorry. Need to confirm your 4A model number, I'm talking a R4AG (R4A Gigabit) version also refered in this title.
2.18.58 maybe for R4AC (100Mbit) only, could not upgrade to 2.28.69.

azimut3, 2022-10-28:
@RadioOperator yes, I have R4AC (100Mbit) and I used 2.18.58