acecilia/OpenWRTInvasion #153: Bus Error after trying to reboot Mi Router 4a R4AC version 2.18.58

Veoxys0910, 2022-10-23:
After that I manually rebooted the router by unplugging it then plugging in it back in. Now the orange light is blinking slowly and then blue light blinks then both lights turn off.

Veoxys0910, 2022-10-23:
BusyBox v1.19.4 (2019-07-30 09:47:57 UTC) built-in shell (ash)
Enter 'help' for a list of built-in commands.

 -----------------------------------------------------
       Welcome to XiaoQiang!
 -----------------------------------------------------
  $$$$$$\  $$$$$$$\  $$$$$$$$\      $$\      $$\        $$$$$$\  $$\   $$\
 $$  __$$\ $$  __$$\ $$  _____|     $$ |     $$ |      $$  __$$\ $$ | $$  |
 $$ /  $$ |$$ |  $$ |$$ |           $$ |     $$ |      $$ /  $$ |$$ |$$  /
 $$$$$$$$ |$$$$$$$  |$$$$$\         $$ |     $$ |      $$ |  $$ |$$$$$  /
 $$  __$$ |$$  __$$< $$  __|        $$ |     $$ |      $$ |  $$ |$$  $$<
 $$ |  $$ |$$ |  $$ |$$ |           $$ |     $$ |      $$ |  $$ |$$ |\$$\
 $$ |  $$ |$$ |  $$ |$$$$$$$$\       $$$$$$$$$  |       $$$$$$  |$$ | \$$\
 \__|  \__|\__|  \__|\________|      \_________/        \______/ \__|  \__|

root@XiaoQiang:~# cd /tmp
root@XiaoQiang:/tmp# curl --insecure https://downloads.openwrt.org/snapshots/targets/ramips/mt76x8/openwrt-ramips-mt76x8
-xiaomi_mi-router-4a-100m-intl-squashfs-sysupgrade.bin --output firmware.bin
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 5632k  100 5632k    0     0   490k      0  0:00:11  0:00:11 --:--:--  953k
root@XiaoQiang:/tmp# ./busybox sha256sum firmware.bin
b88e9873a7a3ba742f8435a609db74ca3ca8d6f9daac3089b82be985ab9d43a9  firmware.bin
root@XiaoQiang:/tmp# mtd -e OS1 -r write firmware.bin OS1
Unlocking OS1 ...
Erasing OS1 ...

Writing from firmware.bin to OS1 ...
Rebooting ...
Bus error

Veoxys0910, 2022-10-23:
BTW I can debrick the router with no problems

azimut3, 2022-10-23:
> After that I manually rebooted the router by unplugging it then plugging in it back in. Now the orange light is blinking slowly and then blue light blinks then both lights turn off then it repeats.

Had the same issue just recently. It happened that I just used the link from the tutorial and after all issue was with the OpenWrt version. When I picked the necessary one everything was installed successfully. I opened an [issue ](https://github.com/acecilia/OpenWRTInvasion/issues/152#issue-1419501831)with some notes. You can check, maybe this can help you.

Veoxys0910, 2022-10-23:
Thank you for the reply @azimut3. I successfully installed OpenWRT on my Mi Router 4a R4AC version 2.18.58 by flashing the MIR4A sysupgrade instead of the R4AC sysupgrade. I still got the Bus Error but after the manual reboot, the router didn't bootloop and OpenWRT was already installed. I got the idea to install the MIR4A sysupgrade from this post https://forum.openwrt.org/t/openwrt-bricks-xiaomi-mi-router-4a-mir4ac-100m-international-version/125759/53

acecilia, 2022-10-23:
This seems solved, closing. Let me know if otherwise 🙌

closed 2022-10-23 by acecilia