acecilia/OpenWRTInvasion #156: [Success] Mi Router 4A Gigabit Edition (R4A) on firmware 3.0.27

Nevinskas, 2022-11-24:
Reporting exploit success on `3.0.27` firmware.
Followed the guide: https://openwrt.org/inbox/toh/xiaomi/xiaomi_mi_router_4a_gigabit_edition

Also, dumped bootloader if ever anyone will need one.

Quick and painless. ;) 
[bootloader_dump.zip](https://github.com/acecilia/OpenWRTInvasion/files/10086943/bootloader_dump.zip)
[exploit.txt](https://github.com/acecilia/OpenWRTInvasion/files/10086945/exploit.txt)
[xiaomi_bootloader_dump.txt](https://github.com/acecilia/OpenWRTInvasion/files/10086946/xiaomi_bootloader_dump.txt)
[xiaomi_dmesg.txt](https://github.com/acecilia/OpenWRTInvasion/files/10086947/xiaomi_dmesg.txt)

closed 2022-12-04 by acecilia

Ged296123, 2023-09-30:
Can confirm this does not work. no connection to router can be established.

acecilia, 2023-10-15:
Usually lack of connection is a problem with your network configuration. There are many issues mentioning the same in the repo, and many of them were solved after changing network configuration, have a look

lavrov-webdev, 2024-08-09:
I can't install software on the router with version 3.0.27 (global) giga. After running the script, I get an error
```bash
Warning: the process has finished, but seems like ssh connection to the router is not working as expected.
* Maybe your firmware version is not supported, please have a look at https://github.com/acecilia/OpenWRTInvasion/blob/master/README.md#unsupported-routers-and-firmware-versions
* Anyway you can try it with: telnet 192.168.31.1
```
I've tried installing it by building on my laptop (MacBook with M chip) and in a Docker container. The error is the same in both cases.

lavrov-webdev, 2024-10-24:
> > I can't install software on the router with version 3.0.27 (global) giga. After running the script, I get an error
> > ```shell
> > Warning: the process has finished, but seems like ssh connection to the router is not working as expected.
> > * Maybe your firmware version is not supported, please have a look at https://github.com/acecilia/OpenWRTInvasion/blob/master/README.md#unsupported-routers-and-firmware-versions
> > * Anyway you can try it with: telnet 192.168.31.1
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > I've tried installing it by building on my laptop (MacBook with M chip) and in a Docker container. The error is the same in both cases.
> 
> Hi, I have the same problem. Did you find any solution?

No, I've already dropped that idea