acecilia/OpenWRTInvasion #158: Warning: the process has finished, but seems like ssh connection to the router is not working as expected.

DannyElfman34, 2022-12-12:
Hello,
First of all, thank you for your work.
I am trying to use the exploit on a Xiaomi Mi4A router, international version, firmware Version 3.0.12
The goal is to flash openwrt.

I have connected my windows PC directly to the Mi router.
I have connected the wan port of the router to my network.
I see the router does have Internet access.

I have built a Debian VM using Virtual box, in network bridge mode to launch the exploit.

Router IP : 192.168.31.1
Windows : 192.168.31.90
VM Debian : 192.168.31.123

And here is what I have after cloning the git repository and launching the script : 

`root@sandbox:/tmp/OpenWRTInvasion# pip3 install -r requirements.txt
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (from -r requirements.txt (line 1)) (2.25.1)
root@sandbox:/tmp/OpenWRTInvasion# python3 remote_command_execution_vulnerability.py
Router IP address [press enter for using the default 'miwifi.com']: 192.168.31.1
Enter router admin password: xxxxxxxxx
There two options to provide the files needed for invasion:
   1. Use a local TCP file server runing on random port to provide files in local directory `script_tools`.
   2. Download needed files from remote github repository. (choose this option only if github is accessable inside router device.)
Which option do you prefer? (default: 1)1
****************
router_ip_address: 192.168.31.1
stok: cc1b49bc0bb312053ae2393b60dcfcfa
file provider: local file server
****************
start uploading config file...
start exec command...
local file server is runing on 0.0.0.0:43369. root='script_tools'
Warning: the process has finished, but seems like ssh connection to the router is not working as expected.
* Maybe your firmware version is not supported, please have a look at https://github.com/acecilia/OpenWRTInvasion/blob/master/README.md#unsupported-routers-and-firmware-versions
* Anyway you can try it with: telnet 192.168.31.1
root@sandbox:/tmp/OpenWRTInvasion# `

The script does not ask me what the stok is (I believe it gets it from the router ip + admin password)

I have the same result using an LCX container from my proxmox server on the same subnet or from my windows pc (with docker/wsl).

Could you please help me because I must say I am a bit lost :(

Thank you

DannyElfman34, 2022-12-13:
Hello,
After all my tests, it appears the script (as mentionned in various posts) is really sensible to the network configuration and does not handle well VM
For the record, I managed to install Openwrt using the exploit via a rpi ethernet connection to the router (where the wan was connected to my main router). 
My rpi was running a minimal debian (octopi server) and everything went smoothly.
By the way, instructions might need an update. No need for the stok anymore as it is pick up via the connection to the router using the admin password.
Thanks again for the exploit !
Cheers all.

closed 2022-12-13 by DannyElfman34

scoresby13, 2024-05-23:
Dear @DannyElfman34 
is there a way to solve this problem without using a rasberry Pi? i don't have one and i am getting the same error message in my terminal "the process has finished, but seems like ssh connection to the router is not working as expected"

silentidiot, 2024-06-12:
got the same error with my mi router 4c. can you teach me how to fix it please?

mhasandx, 2024-08-08:
> got the same error with my mi router 4c. can you teach me how to fix it please?

Did u found any solutions ?? i am currently facing the same problem on my mi router 4c

michalloooo, 2025-07-16:
I plugged in the network cable only during the initial setup, when the system required an internet connection. Right after that, I unplugged the cable, and the following steps worked without any issues. Let me know if it works for you.