acecilia/OpenWRTInvasion #165: [Success] Mi Router 4A 100M on firmware 3.0.12

SilentoA, 2023-01-26:
Hello! I wanted to thank you very much for OpenWRTInvasion and report back on the success!

Device: MiRouter **4A 100M (non gigabit)**
Software version: **3.0.12**

The process of getting root:

```
------------------------------------------------------------------------------------------------------------------------------------------------------------------------
~/OpenWRTInvasion (master) » python3 remote_command_execution_vulnerability.py                                                                     130 ↵ liveuser@ctlos
Router IP address [press enter for using the default 'miwifi.com']: 192.168.31.1           
Enter router admin password: 0-)3LJIg|D=Pl=z2(WwI1{-9d
There two options to provide the files needed for invasion:
   1. Use a local TCP file server runing on random port to provide files in local directory `script_tools`.
   2. Download needed files from remote github repository. (choose this option only if github is accessable inside router device.)
Which option do you prefer? (default: 1)
****************
router_ip_address: 192.168.31.1
stok: c42bd637f2f363439c19af8b006d6f47
file provider: local file server
****************
start uploading config file...
start exec command...
local file server is runing on 0.0.0.0:47791. root='script_tools'
local file server is getting 'busybox-mipsel' for 192.168.31.1.
local file server is getting 'dropbearStaticMipsel.tar.bz2' for 192.168.31.1.
done! Now you can connect to the router using several options: (user: root, password: root)
* telnet 192.168.31.1
* ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -oHostKeyAlgorithms=+ssh-rsa -c 3des-cbc -o UserKnownHostsFile=/dev/null root@192.168.31.1
* ftp: using a program like cyberduck
------------------------------------------------------------------------------------------------------------------------------------------------------------------------
```

OpenWrt installation process:

```
------------------------------------------------------------------------------------------------------------------------------------------------------------------------
~ » ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -oHostKeyAlgorithms=+ssh-rsa -c 3des-cbc -o UserKnownHostsFile=/dev/null root@192.168.31.1           liveuser@ctlos
The authenticity of host '192.168.31.1 (192.168.31.1)' can't be established.
RSA key fingerprint is SHA256:cGn3yDg2gfyMoGIh1pKGxWDWZWiHK1vj6/S9iRlljlo.
This key is not known by any other names.
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added '192.168.31.1' (RSA) to the list of known hosts.
root@192.168.31.1's password: 

BusyBox v1.19.4 (2020-12-22 12:08:23 UTC) built-in shell (ash)
Enter 'help' for a list of built-in commands.

 -----------------------------------------------------
       Welcome to XiaoQiang!
 -----------------------------------------------------
  $$$$$$\  $$$$$$$\  $$$$$$$$\      $$\      $$\        $$$$$$\  $$\   $$\
 $$  __$$\ $$  __$$\ $$  _____|     $$ |     $$ |      $$  __$$\ $$ | $$  |
 $$ /  $$ |$$ |  $$ |$$ |           $$ |     $$ |      $$ /  $$ |$$ |$$  /
 $$$$$$$$ |$$$$$$$  |$$$$$\         $$ |     $$ |      $$ |  $$ |$$$$$  /
 $$  __$$ |$$  __$$< $$  __|        $$ |     $$ |      $$ |  $$ |$$  $$<
 $$ |  $$ |$$ |  $$ |$$ |           $$ |     $$ |      $$ |  $$ |$$ |\$$\
 $$ |  $$ |$$ |  $$ |$$$$$$$$\       $$$$$$$$$  |       $$$$$$  |$$ | \$$\
 \__|  \__|\__|  \__|\________|      \_________/        \______/ \__|  \__|

root@XiaoQiang:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 01000000 00010000 "ALL"
mtd1: 00020000 00010000 "Bootloader"
mtd2: 00010000 00010000 "Config"
mtd3: 00010000 00010000 "Factory"
mtd4: 00010000 00010000 "crash"
mtd5: 00010000 00010000 "cfg_bak"
mtd6: 00200000 00010000 "overlay"
mtd7: 00da0000 00010000 "OS1"
[... 75 lines cut ...]
-rw-r--r--    1 root     root             0 Jan 25 11:59 upnp.leases
-rw-r--r--    1 root     root             0 Jan 23 08:47 web_config_list
prw-------    1 root     root             0 Jan 26 06:12 web_filter_list
-rw-------    1 root     root         55153 Jan 26 06:01 wifi_analysis.log
root@XiaoQiang:/tmp# mtd -r write openwrt.bin OS1
Unlocking OS1 ...

Writing from openwrt.bin to OS1 ...     
Rebooting ...

```

The next step was to install luci:

```
ssh root@192.168.1.1
opkg update
opkg install luci
```

Then you can get into the web interface.

[image]

Please update information about supported firmware for MiRouter 4A 100M (non gigabit). Thanks again!

acecilia, 2023-01-27:
Awesome thanks!

closed 2023-01-27 by acecilia

AsoTora, 2023-03-23:
I had the exact input parameters: MiRouter 4A 100M (R4AC) International, Software version: 3.0.12, didn't work.

DanilBorchevkin, 2023-04-11:
Evironment:
1. MiRouter 4A 100M (R4AC) International
2. Firmware version: 3.0.12
3. Windows 10

I tried to getting things done using
1. docker solution with different combinations of options (local files / remote files)
2. python host solutions with different combinations (local files / remote files)

and it doesn't works

Log
```
(base) PS C:\workspace\docker\OpenWRTInvasion> docker run --network host -it openwrtinvasion
Router IP address [press enter for using the default 'miwifi.com']: 192.168.31.1
Enter router admin password: 009452qwe
There two options to provide the files needed for invasion:
   1. Use a local TCP file server runing on random port to provide files in local directory `script_tools`.
   2. Download needed files from remote github repository. (choose this option only if github is accessable inside router device.)
Which option do you prefer? (default: 1)
****************
router_ip_address: 192.168.31.1
stok: 737fd60b3febe56cf92d2c52359763f4
file provider: local file server
****************
start uploading config file...
start exec command...
local file server is runing on 0.0.0.0:60383. root='script_tools'
Warning: the process has finished, but seems like ssh connection to the router is not working as expected.
* Maybe your firmware version is not supported, please have a look at https://github.com/acecilia/OpenWRTInvasion/blob/master/README.md#unsupported-routers-and-firmware-versions
```

Olejja, 2023-06-03:
Thank you! Successfully flashed.

EndermanchFan2100, 2023-10-20:
Where can i get the global firmware?

Olejja, 2023-10-23:
By global, you mean international?

Here is a snapshot for R4AC >> https://openwrt.org/inbox/toh/xiaomi/r4ac

Here is the latest release (I haven't tested it) >> https://downloads.openwrt.org/releases/23.05.0/targets/ramips/mt76x8/

Olejja, 2023-10-23:
> Where can i get the global firmware?

By global, you mean international?

Here is a snapshot for R4AC >> https://openwrt.org/inbox/toh/xiaomi/r4ac

Here is the latest release (I haven't tested it) >> https://downloads.openwrt.org/releases/23.05.0/targets/ramips/mt76x8/

Olejja, 2023-10-23:
> Yeah, international but i need stock that is on english, not chinese.

Sorry, but you couldn’t specify what you need, otherwise I don’t quite understand you...

anupdebnath, 2023-10-23:
> > Yeah, international but i need stock that is on english, not chinese.
> 
> Sorry, but you couldn’t specify what you need, otherwise I don’t quite understand you...

He is asking for stock firmware in English. 3.xx.xx

Olejja, 2023-10-24:
> He is asking for stock firmware in English. 3.xx.xx

Now it's clear, thank you.

@EndermanchFan2100 I found [3.0.5](https://forum.openwrt.org/t/xiaomi-r4ac-problems-installing-openwrt/82083/3) and [3.0.10](https://4pda.to/forum/index.php?showtopic=975951&st=1660#entry101050944) global. Be careful.

EndermanchFan2100, 2023-10-24:
> > He is asking for stock firmware in English. 3.xx.xx
> 
> Now it's clear, thank you.
> 
> @EndermanchFan2100 I found [3.0.5](https://forum.openwrt.org/t/xiaomi-r4ac-problems-installing-openwrt/82083/3) and [3.0.10](https://4pda.to/forum/index.php?showtopic=975951&st=1660#entry101050944) global. Be careful.

I have also found both of those links but both of them are expired.

Olejja, 2023-10-25:
> I have also found both of those links but both of them are expired.

I had to register to download the firmware for you from the second link.
[miwifi_r4ac_firmware_0942f_3.0.10_INT.zip](https://github.com/acecilia/OpenWRTInvasion/files/13161726/miwifi_r4ac_firmware_0942f_3.0.10_INT.zip)

EndermanchFan2100, 2023-10-25:
> > I have also found both of those links but both of them are expired.
> 
> I had to register to download the firmware for you from the second link. [miwifi_r4ac_firmware_0942f_3.0.10_INT.zip](https://github.com/acecilia/OpenWRTInvasion/files/13161726/miwifi_r4ac_firmware_0942f_3.0.10_INT.zip)
> 
> Also read [this comment](https://4pda.to/forum/index.php?showtopic=975951&st=1680#entry101053438) before you start converting the Chinese version into a global one. Good luck.

Thank you.

gnssefa, 2023-12-20:
I NEED A HELP PLEASE !
[image]

sepehr-72, 2024-04-05:
> I had to register to download the firmware for you from the second link. [...]

Hey sir , kind regards . I have MiRouter 4A ( not giga ) and i think it's Chinese but the menu is English . on version 3.0.10 , can i install openwrt? Just yes or no

Please accept my apologies for poor English ❤️♥️

anupdebnath, 2024-04-05:
> but the menu is English

yes, you can install OpenWrt.

sepehr-72, 2024-04-06:
> > but the menu is English
> 
> yes, you can install OpenWrt.

Thank you so much , i really appreciate it .

 **is there any youtube guide ?**

testmanavr, 2024-09-08:
Hello, I need help
I have Xiaomi Mi Router 4A v2 (R4ACv2) as well.
I firstly installed openwrt from mi firmware version 3.0.129.
I then installed openwrt for the 100m international (v1 i guess)
then I found a version compiled version for v2 and now it's blinking orange. Please help recovering it.
Thanks a lot

dharenkamp, 2024-09-16:
Have successfully flashed openwrt snapshot to R4ACv2.
Device comes with fw version 3.0.129.

invasion was successful:
`(venv) xxxxxxxxx@xxxxxxxxxxxx:~/temp/penv/OpenWRTInvasion$ python3 remote_command_execution_vulnerability.py
Router IP address [press enter for using the default 'miwifi.com']: 
Enter router admin password: 123456789
There two options to provide the files needed for invasion:
   1. Use a local TCP file server runing on random port to provide files in local directory `script_tools`.
   2. Download needed files from remote github repository. (choose this option only if github is accessable inside router device.)
Which option do you prefer? (default: 1)
****************
router_ip_address: miwifi.com
stok: 8991b8a2a300b366ebeef79ef4a035bf
file provider: local file server
****************
start uploading config file...
start exec command...
local file server is runing on 0.0.0.0:49509. root='script_tools'
local file server is getting 'busybox-mipsel' for 192.168.31.1.
local file server is getting 'dropbearStaticMipsel.tar.bz2' for 192.168.31.1.
done! Now you can connect to the router using several options: (user: root, password: root)
* telnet miwifi.com
* ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -oHostKeyAlgorithms=+ssh-rsa -c 3des-cbc -o UserKnownHostsFile=/dev/null root@miwifi.com
* ftp: using a program like cyberduck`

WIth Openwrt FIrmware Selector https://firmware-selector.openwrt.org i´ve get the image from snapshot for the "Xiaomi Mi Router 4A 100M International Edition V2" device.

`(venv) dharenkamp@denbg-mac-pp001:~/temp/penv/OpenWRTInvasion$ ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -oHostKeyAlgorithms=+ssh-rsa -c 3des-cbc -o UserKnownHostsFile=/dev/null root@miwifi.com
The authenticity of host 'miwifi.com (192.168.31.1)' can't be established.
RSA key fingerprint is SHA256:lyxLvDJy+Dqh6fEqgSj0xXIXWPbHUW35oJf5e4cUsuc.
This key is not known by any other names
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added 'miwifi.com' (RSA) to the list of known hosts.
root@miwifi.com's password: 

BusyBox v1.19.4 (2022-09-14 13:16:45 UTC) built-in shell (ash)
Enter 'help' for a list of built-in commands.

 -----------------------------------------------------
       Welcome to XiaoQiang!
 -----------------------------------------------------
  $$$$$$\  $$$$$$$\  $$$$$$$$\      $$\      $$\        $$$$$$\  $$\   $$\
 $$  __$$\ $$  __$$\ $$  _____|     $$ |     $$ |      $$  __$$\ $$ | $$  |
 $$ /  $$ |$$ |  $$ |$$ |           $$ |     $$ |      $$ /  $$ |$$ |$$  /
 $$$$$$$$ |$$$$$$$  |$$$$$\         $$ |     $$ |      $$ |  $$ |$$$$$  /
 $$  __$$ |$$  __$$< $$  __|        $$ |     $$ |      $$ |  $$ |$$  $$<
 $$ |  $$ |$$ |  $$ |$$ |           $$ |     $$ |      $$ |  $$ |$$ |\$$\
 $$ |  $$ |$$ |  $$ |$$$$$$$$\       $$$$$$$$$  |       $$$$$$  |$$ | \$$\
 \__|  \__|\__|  \__|\________|      \_________/        \______/ \__|  \__|

root@XiaoQiang:~# cd /tmp/
root@XiaoQiang:/tmp# mv openwrt-236a0ee5829b-ramips-mt76x8-xiaomi_mi-router-4a-100m-intl-v2-squashfs-sysupgrade.bin firmware.bin
root@XiaoQiang:/tmp# mtd -e OS1 -r write firmware.bin OS1
Unlocking OS1 ...
Erasing OS1 ...

Writing from firmware.bin to OS1 ...     
Rebooting ...
`

So openwrt install was successful.

aymenmed0001, 2024-09-19:
you can get into the web interface for xiaomi 4a v2 with openwrt and thankyou @([dharenkamp](https://github.com/dharenkamp))

cfercl, 2025-11-01:
> Thank you.
@EndermanchFan2100 
Hi friend!  Were you able to revert to the international stock version from the version?

cfercl, 2025-11-04:
> Hello does anyone have the English firmware for Xiaomi router 4a R4AC? because when I install this miwifi_r4ac_firmware_0942f_3.0.10_INT.zip the router bricked

Hello, if you have a china-version installed there is no a easy way to back to INT version, I had this problem but..
I followed this steps sucessfully -> https://4pda.to/forum/index.php?showtopic=975951&view=findpost&p=101053438

I had to use google translator, created an account,  install python, etc  but everything ran well. (Sorry for my english)