acecilia/OpenWRTInvasion #166: Hoddys Bricked video has malware-infested tools

juancarlospdt, 2023-01-29:
Just got bricked, i got into Hoddys video and downloaded the tools. As i usually do i took the tool to VirusTotal and the web turned on red over the malware it brings.

Just talking over the most referential security web on the industry and malware not related to modifying files(like could be KMS) malware to DIRECTLY STEAL data.

Despite the video is really aproachable, I would suggest deleting it until the tools get cleaned up...

acecilia, 2023-03-13:
@hoddy Anything to mention?

hoddy, 2023-03-13:
I'm assuming this is about the debrick tool rather than OpenWRTInvasion, it's just a copy of TinyPXE straight from their website with the firmware renamed as test.bin in its root folder. I can only assume this is a false positive, probably because of the way it needs network and file access to serve the file over TFTP. Anyone concerned can make their own tools I even explain how to make your own on the page where the tools are kept. If anyone has any other suggestions happy make changes if required?

acecilia, 2023-03-13:
Cool, thanks for clarification. Closing, feel free to reopen of you do not agree @juancarlospdt

closed 2023-03-13 by acecilia