acecilia/OpenWRTInvasion #169: where can i find the indian rom for this router?

XDAGamer, 2023-02-12:
so i bricked my router in process of forcing openwrt through telnet though it was not supported according to this github repository. Now the thing is im on on global rom and want the indian unit firmware, where can i find it?

XDAGamer, 2023-02-12:
i have the gigabit version, v2 i think

410252889, 2023-03-04:
https://github.com/acecilia/OpenWRTInvasion/files/9920159/miwifi_r4av2_all_release_2.30.25.bin.zip

XDAGamer, 2023-03-11:
not working actually i was mistaken only global version is working

closed 2023-03-13 by acecilia

XDAGamer, 2023-03-13:
why has this been closed? i am not able to get the rom working

acecilia, 2023-03-13:
Because this issue has no relation with this repository. This repository is hosting a tool to get access to the OS of the router. It is not hosting ROMs

XDAGamer, 2023-03-17:
i might have v1 because this firmware is not working yet chinese and global are fine which both are v1 in my reckoning because it has similar name format but only thing they don't do is mention v2 in their name like this one