acecilia/OpenWRTInvasion #175: cannot flash Mi Router 4A (non gigabit) | MiWiFi Release 3.0.10

PAOK1926, 2023-04-28:
i have a Mi Router 4A (non gigabit) | MiWiFi Release 3.0.10 | Country Greece

OpenWRTInvasion-master % python3 remote_command_execution_vulnerability.py
Router IP address [press enter for using the default 'miwifi.com']: 192.168.31.1            
Enter router admin password: 22222222
There two options to provide the files needed for invasion:
   1. Use a local TCP file server runing on random port to provide files in local directory `script_tools`.
   2. Download needed files from remote github repository. (choose this option only if github is accessable inside router device.)
Which option do you prefer? (default: 1)
****************
router_ip_address: 192.168.31.1
stok: b7c2b50a3973817c0934b5291f2eee29
file provider: local file server
****************
start uploading config file...
start exec command...
local file server is runing on 0.0.0.0:63088. root='script_tools'
Warning: the process has finished, but seems like ssh connection to the router is not working as expected.
* Maybe your firmware version is not supported, please have a look at https://github.com/acecilia/OpenWRTInvasion/blob/master/README.md#unsupported-routers-and-firmware-versions
* Anyway you can try it with: telnet 192.168.31.1

OpenWRTInvasion-master % telnet 192.168.31.1
Trying 192.168.31.1...
telnet: connect to address 192.168.31.1: Connection refused
telnet: Unable to connect to remote host

jtinbergen, 2023-11-16:
I have the same problem. Connected directly to device, run invasion.. but process did not finish correctly.  Running R4A with 3.0.24.

alfadex, 2024-01-07:
same here with version MiWiFi Release 3.0.12, I tried both ubuntu and fedora. I have tired with it.i'll leave it with stock